robotology/yarp #2768: Since mid october 2021 master branch is producing malformed C++ if thrift file contain comments with double quotes

traversaro, 2021-11-23:
**Describe the bug**
Since mid october 2021, several downstream projects that contained double quotes failed when compiled with YARP 3.5.100 with errors like:
~~~
Consolidate compiler generated dependencies of target idyntree-yarp-visualizer
[ 29%] Building CXX object src/modules/idyntree-yarp-visualizer/CMakeFiles/idyntree-yarp-visualizer.dir/thrifts/VisualizerCommands.cpp.o
/home/traversaro/test/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:283:43: error: stray ‘\’ in program
  283 |         "@return A status message. \\"[ok]\\" in case of success"
      |                                           ^
/home/traversaro/test/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:283:44: error: stray ‘\’ in program
  283 |         "@return A status message. \\"[ok]\\" in case of success"
      |                                            ^
/home/traversaro/test/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:283:40: error: ‘ok’ was not declared in this scope
  283 |         "@return A status message. \\"[ok]\\" in case of success"
      |                                        ^~
/home/traversaro/test/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:283:45: error: expected ‘}’ before string constant
  283 |         "@return A status message. \\"[ok]\\" in case of success"
      |                                             ^~~~~~~~~~~~~~~~~~~~~
/home/traversaro/test/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:281:40: note: to match this ‘{’
  281 |     static constexpr const char* s_help{
      |                                        ^
/home/traversaro/test/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:283:42: error: expected ‘;’ at end of member declaration
  283 |         "@return A status message. \\"[ok]\\" in case of success"
      |                                          ^
      |                                           ;
/home/traversaro/test/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:283:45: error: expected unqualified-id before string constant
  283 |         "@return A status message. \\"[ok]\\" in case of success"
      |                                             ^~~~~~~~~~~~~~~~~~~~~
/home/traversaro/test/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:285:1: error: expected declaration before ‘}’ token
  285 | };
      | ^
make[5]: *** [src/modules/idyntree-yarp-visualizer/CMakeFiles/idyntree-yarp-visualizer.dir/build.make:126: src/modules/idyntree-yarp-visualizer/CMakeFiles/idyntree-yarp-visualizer.dir/thrifts/VisualizerCommands.cpp.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:377: src/modules/idyntree-yarp-visualizer/CMakeFiles/idyntree-yarp-visualizer.dir/all] Error 2
make[3]: *** [Makefile:136: all] Error 2
make[2]: *** [CMakeFiles/idyntree-yarp-tools.dir/build.make:85: src/idyntree-yarp-tools/CMakeFiles/YCMStamp/idyntree-yarp-tools-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:1819: CMakeFiles/idyntree-yarp-tools.dir/all] Error 2
make: *** [Makefile:101: all] Error 2
~~~

**To Reproduce**
Steps to reproduce the behavior:

Try to configure icub-main `v1.21.0` with YARP 3.5.100 to reproduce the problem. The problem was workarounded in the latest master in https://github.com/robotology/icub-main/pull/767 .

**Expected behavior**
The file generated by thrift should be well-formed.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Configuration (please complete the following information):**
 - OS:
 - yarp version: 3.5.100
-  compiler: gcc 9

**Additional context**
See https://github.com/robotology/robotology-superbuild/issues/900 . We can also close this issue if this is the intended behavior, I wanted just to have a placeholder/reference for this problem.

traversaro, 2021-11-23:
Back in october, we identified that probably this was a regression due to https://github.com/robotology/yarp/pull/2709 .

drdanz, 2021-12-02:
This is the commit causing the issue: d4c5e742ea7d00ea73758ddbdac1eaab368d56f0

PR #2771 (2021-12-03): yarpidl_thrift: Fix malformed files when comments contain double quotes
Fixes #2768
commit 0c898e33 (2021-12-03): yarpidl_thrift: Fix malformed files when comments contain double quotes

Fixes #2768
closed 2021-12-03 by drdanz