robotology/yarp #2769: Yarpscope not working properly on MacOS with multiple coloumns/row

lrapetti, 2021-11-25:
**Describe the bug**
When running a `yarpscope` with multiple rows/coloumns, only one of the plots is shown correctly. The behaviour also seems to be changing if I move the scope between the two screens connected to my laptop (see the following video). (Note that the problem is happening even if no external screen is connected)

https://user-images.githubusercontent.com/35487806/143460300-0ed6bc9e-8c4a-4254-b35b-7af998b7a5f5.mp4

**To Reproduce**
Here are the configuration files that I have tested in the video below:
- multiple columns: [yarpscope_test.xml.txt](https://github.com/robotology/yarp/files/7603695/yarpscope_test.xml.txt)
- multiple rows: [yarpscope_test_2.xml.txt](https://github.com/robotology/yarp/files/7603699/yarpscope_test_2.xml.txt)

I run then without the source data being published, but I still would expect to see the plots (with data=0).
```
$ yarpscope --xml yarpscope_test.xml
```

**Expected behavior**
I would expect to visualize the two plots, as I see it working on windows:
- MacOS
  [image]
- Windows
  [image]

**Configuration:**
 - OS: MacOS 11.3.1
 - yarp version: 3.4.6
-  compiler: clang version 12.0.5

**Additional context**
I think a similar problem was experienced in Ubuntu. @kouroshD can you confirm it?

kouroshD, 2021-11-25:
> I think a similar problem was experienced in Ubuntu. @kouroshD can you confirm it?

Yes, that's correct.

drdanz, 2021-12-02:
This is probably a bug in Qt5. What version are you using?

lrapetti, 2021-12-02:
> This is probably a bug in Qt5. What version are you using?

`5.15.2`