robotology/yarp #2798: Add timeout parameter to yarp detect and yarp where

Nicogene, 2022-01-11:
**Is your feature request related to a problem? Please describe.**
When you do `yarp detect` or `yarp where` and in your `.conf` file you have an unreachable IP, it get stuck forever, because it is not possible to set `timeout` argument to the `Contact` and the default timeout is `-1`:

https://github.com/robotology/yarp/blob/ce6f008c86ba58a652b04b6bffdc2b46e9b157e2/src/libYARP_os/src/yarp/os/Contact.cpp#L60

This problem can be reproduced as:
```
yarp conf 1.2.3.4 10000
```

And then it get stuck like this:
```
Looking for name server on 1.2.3.4, port number 10000
If there is a long delay, try:
  yarp conf --clean

```

This is particular annoying if you use this commands in scripts because you have to stop everything, run `yarp conf --clean` and then restart the scripts again.

**Describe the solution you'd like**
Add a `timeout` argument for the two companion commands

Nicogene, 2022-01-11:
If you agree on this solution I can open the PR by myself

randaz81, 2022-04-12:
I tried your test case with ip 1.2.3.4, but i cannot reproduce it. Both yarp detect and where exit correctly with a messagr after few seconds.
Maybe there is something more that I dont know?

randaz81, 2022-04-21:
I wonder if you were using some virtualized environment or WSL or something else which may block the multicast packets sent to find if a server is running on the network. (just an hypothesis, because I cannot reproduce your issue)

Nicogene, 2022-04-21:
> I tried your test case with ip 1.2.3.4, but i cannot reproduce it. Both yarp detect and where exit correctly with a messagr after few seconds. Maybe there is something more that I dont know?

Sorry for the late reply! If I remember correctly this issue was on the laptop of the robot (then a Ubuntu 20.04) but I can double-check