robotology/yarp #2807: yarpviz crashes with segmentation fault

lrapetti, 2022-02-21:
**Describe the bug**
We have tried to use `yarpviz`, but after launching `analyze` command. It start with the analysis but at a certain point it crashed with segmentation fault.

Here are the final prints in the log:
```
...
[INFO] |yarp.os.impl.PortCoreOutputUnit|/yarpviz| Sending output from /yarpviz to /yarpmotorgui0//icub/torso/stateExt:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/yarpviz| Removing output from /yarpviz to /yarpmotorgui0//icub/torso/stateExt:i
[WARNING] Found a nonexistent port ( null ) in the output list of /icub/cam/left
[WARNING] Found a nonexistent port ( null ) in the output list of /icub/cam/right
[WARNING] Found a nonexistent port ( null ) in the output list of /icub/camcalib/left/out
[WARNING] Found a nonexistent port ( null ) in the output list of /icub/camcalib/right/out
[WARNING] Found a nonexistent port ( /idyntree-yarp-visualizer/remoteControlBoard:i/icub/head/stateExt:i ) in the output list of /icub/head/stateExt:o
Segmentation fault (core dumped)
```

**Configuration (please complete the following information):**
 - OS: Ubuntu 20.04.3
 - yarp version: 3.6.0

cc @S-Dafarra @RiccardoGrieco

traversaro, 2022-02-21:
I wonder if this is related to the `Found a nonexistent port ( null )`, perhaps this triggers a code path that eventually leads to seg fault.

randaz81, 2022-05-24:
I cannot reproduce your error. Can you provide more details on how to get a 'null' port? (sharing the application .xml it would be great, if possible)

S-Dafarra, 2022-05-24:
Those ports were related to the eyes. The corresponding application was https://github.com/robotology/walking-teleoperation/blob/916ae1bb63f610c21d95bd62391908884a9d8385/app/scripts/Xprize-VisualRetargeting.xml

I wonder if it has to do with the ``mjpeg`` carrier.

randaz81, 2022-05-24:
mjpeg carrier operates in a very peculiar behavior: you are not sending data from src to dest. It's the dest that requires data to be sent from the src.
I'm not sure if this might be the issue, but I'll check.

randaz81, 2022-05-24:
Regarding the `null` port:
I tested it using  just the `yarp ping` command and I can confirm the theory of the reverse logic of the mjpeg carrier:
```
yarp ping /yarpview/img:i
This is "/yarpview/img:i" at "tcp://192.168.232.205:10006/"
There are no outgoing connections
There is an input connection from "/grabber" to "/yarpview/img:i" using mjpeg
There is an input connection from "<ping>" to "/yarpview/img:i" using text_ack

yarp ping /grabber
This is "/grabber" at "tcp://192.168.232.205:10003/"
There is an output connection from "/grabber" to [null] using mjpeg
There is an input connection from "<ping>" to "/grabber" using text_ack
```
However, at very first glance, this does not seem associated with the segfault issue we are discussing here (I can run yarpviz with no problems), and I'd discuss this issue separately (it can be fixed), if you tell me that yarpviz works when the cameras' ports are not connected.

Which version of the graphviz library are you using, btw? I'm using version 2.50 on Windows, which is not the latest one available, which is 3.0, and was causing me some linking errors (maybe also this one is a different issue)

S-Dafarra, 2022-05-24:
> However, at very first glance, this does not seem associated with the segfault issue we are discussing here (I can run yarpviz with no problems), and I'd discuss this issue separately (it can be fixed), if you tell me that yarpviz works when the cameras' ports are not connected.

If I remember correctly, we tried also without those connections, but it was still segfaulting. We were not sure about the possible cause. We just guessed it might have been related to the ``null`` port because that was the last message printed.

I guess it is related to GUI, not on how the ports are retrieved. Since we were curious to print all the connections, I wrote a small companion to run a simple test in https://github.com/S-Dafarra/yarp/tree/yarp_name_connections

<details>
<summary>The output is here </summary>

<p>

```
==================== Connections ================================
(yarprobotinterface@10.0.0.2)/wholeBodyDynamics/right_arm/cartesianEndEffectorWrench:o --(fast_tcp)--> (WeightDisplayModule@10.0.10.164)/WeightDisplay/right_hand:i
(yarprobotinterface@10.0.0.2)/wholeBodyDynamics/left_arm/cartesianEndEffectorWrench:o --(fast_tcp)--> (WeightDisplayModule@10.0.10.164)/WeightDisplay/left_hand:i
(yarprobotinterface@10.0.10.164)/iCub/RobotStateWrapper/state:o --(tcp)--> (HumanStateVisualizer@10.0.10.164)/tmp/port/13(yarprobotinterface@10.0.10.164)/iCub/RobotStateWrapper/state:o --(fast_tcp)--> (XsensRetargetingModule@10.0.0.158)/iFeelRetargeting/HumanStateWrapper/state:i
(WalkingModule@10.0.0.2)/walking-coordinator/torsoYaw:o --(fast_tcp)--> (VirtualizerModule.exe@10.0.10.120)/virtualizer/robotOrientation:i
(WeightDisplayModule@10.0.10.164)/WeightDisplay/out:o --(fast_tcp)--> (yarpdev.exe@10.0.10.120)/joypadDevice/Oculus/label_1
(VirtualizerModule.exe@10.0.10.120)/virtualizer/goal:o --(fast_tcp)--> (WalkingModule@10.0.0.2)/walking-coordinator/goal:i
(SRanipalModule.exe@10.0.10.120)/SRanipalModule/emotions:o --(fast_tcp)--> (emotionInterface@10.0.0.2)/icub/face/emotions/in
(yarpdev.exe@10.0.10.120)/icub-virtualizer/microphone/audio:o --(fast_tcp)--> (yarpdev@10.0.0.2)/icub/speakers/audio:i
(idyntree-yarp-visualizer@10.0.0.158)/idyntree-yarp-visualizer/image --(fast_tcp)--> (yarpdev.exe@10.0.10.120)/joypadDevice/Oculus/gui_0
(SRanipalModule.exe@10.0.10.120)/SRanipalModule/VR/rpc:o --(fast_tcp)--> (yarpdev.exe@10.0.10.120)/joypadDevice/Oculus/rpc
(yarpdev@10.0.0.2)/icub/microphone/audio:o --(fast_tcp)--> (yarpdev.exe@10.0.10.120)/icub-virtualizer/speakers/audio:i
(yarpdev@10.0.0.2)/icub/microphone/audio:o --(fast_tcp)--> (yarpdev.exe@10.0.10.120)/icub-virtualizer/speakers/audio:i(yarpdev@10.0.0.2)/icub/microphone/audio:o --(fast_tcp)--> (yarpdev@10.0.10.164)/icub-console-gui/speakers/audio:i
(WeightRetargetingModule@10.0.10.164)/WeightRetargeting/output:o --(fast_tcp)--> (yarprobotinterface@10.0.10.164)/iFeelSuit/WearableActuatorsCommand/input:i
(XsensRetargetingModule@10.0.0.158)/iFeelRetargeting/jointPosition:o --(fast_tcp)--> (WalkingModule@10.0.0.2)/walking-coordinator/jointPosition:i
(yarprobotinterface@10.0.0.2)/icub/skin/right_hand --(udp)--> (HapticGloveModule@10.0.0.2)/icub/skin/right_hand/in(yarprobotinterface@10.0.0.2)/icub/skin/right_hand --(fast_tcp)--> (iCubSkinGui@10.0.10.164)/skinGui/right_hand:i
(yarpdev@10.0.10.164)/icub-console/operator_microphone/audio:o --(fast_tcp)--> (yarpdev.exe@10.0.10.120)/icub-virtualizer/speakers_operator/audio:i
(yarprobotinterface@10.0.0.2)/icub/skin/left_hand --(udp)--> (HapticGloveModule@10.0.0.2)/icub/skin/left_hand/in(yarprobotinterface@10.0.0.2)/icub/skin/left_hand --(fast_tcp)--> (iCubSkinGui@10.0.10.164)/skinGui/left_hand:i
(camCalib@10.0.0.158)/icub/camcalib/right/out --(mjpeg)--> (@)
(yarpdev@10.0.0.2)/icub/cam/right --(mjpeg)--> (@)
(yarpdev@10.0.0.2)/icub/cam/left --(mjpeg)--> (@)
(camCalib@10.0.0.158)/icub/camcalib/left/out --(mjpeg)--> (@)
(yarpmotorgui@10.0.0.158)/yarpmotorgui0//icub/right_leg/rpc:o --(tcp)--> (yarprobotinterface@10.0.0.2)/icub/right_leg/rpc:i
(yarpmotorgui@10.0.0.158)/yarpmotorgui0//icub/right_arm/rpc:o --(tcp)--> (yarprobotinterface@10.0.0.2)/icub/right_arm/rpc:i
(yarpmotorgui@10.0.0.158)/yarpmotorgui0//icub/left_leg/rpc:o --(tcp)--> (yarprobotinterface@10.0.0.2)/icub/left_leg/rpc:i
(yarpmotorgui@10.0.0.158)/yarpmotorgui0//icub/head/rpc:o --(tcp)--> (@)
(yarpmotorgui@10.0.0.158)/yarpmotorgui0//icub/face/rpc:o --(tcp)--> (yarprobotinterface@10.0.0.2)/icub/face/rpc:i
(yarprobotinterface@10.0.0.158)/yarp_robot_logger/xsens_inertial/rpc:i --(tcp)--> (yarprobotinterface@10.0.0.2)/icub/head/inertials/rpc:o
(yarprobotinterface@10.0.0.158)/yarp_robot_logger/right_rear_ft/rpc:o --(tcp)--> (yarprobotinterface@10.0.0.2)/icub/right_foot_heel/analog:o/rpc:i
(yarprobotinterface@10.0.0.158)/yarp_robot_logger/joints/icub/torso/rpc:o --(tcp)--> (yarprobotinterface@10.0.0.2)/icub/torso/rpc:i
(yarprobotinterface@10.0.0.2)/wholeBodyDynamics/right_foot_front/cartesianEndEffectorWrench:o --(udp)--> (yarprobotinterface@10.0.0.158)/yarp_robot_logger/right_front_wrench(yarprobotinterface@10.0.0.2)/wholeBodyDynamics/right_foot_front/cartesianEndEffectorWrench:o --(tcp)--> (WalkingModule@10.0.0.2)/walking-coordinator/right_foot_front/cartesianEndEffectorWrench:i
(yarprobotinterface@10.0.0.2)/wholeBodyDynamics/netExternalWrenches:o --(tcp)--> (idyntree-yarp-visualizer@10.0.0.158)/idyntree-yarp-visualizer/netExternalWrenches:i
[... cut ...] rm/stateExt:o --(udp)--> (WalkingModule@10.0.0.2)/walking-coordinator/remoteControlBoard/icub/right_arm/stateExt:i
```

</p>

</details>

> Which version of the graphviz library are you using, btw? I'm using version 2.50 on Windows, which is not the latest one available, which is 3.0, and was causing me some linking errors (maybe also this one is a different issue)

We installed it from ``apt`` in Ubuntu. The version should be 2.42.2-3build2, see https://packages.ubuntu.com/focal/libgraphviz-dev