robotology/yarp #2812: robotinterface: it is not possible to pass a list with extern-name

lrapetti, 2022-03-09:
**Describe the bug**
I was trying to use the `extern-name` option to pass the value of a parameter for a `yarprobotinterface` via command line.
In particular, the option I am interested in is a vector of double, indeed I was flagging the option as `extern-name`:
```xml
...
<param name="dynamicalIKCorrectionGainsLinRot" extern-name="dIKGains">(8.0 20.0)</param>
...
```
and then launching the yarpmanager application as follow:
```sh
$ yarprobotinterface --config my_application --dIKGains "(10.0,100.0)"
```
however, it seems that the resulting configuration is not a list, but rather a simple value. this is what I see if I try to print the configuration as string:
```c
...
std::cerr << "dynamicalIKCorrectionGainsLinRot is " << config.find("dynamicalIKCorrectionGainsLinRot").toString() << std::endl;
...
```
```sh
dynamicalIKCorrectionGainsLinRot is 10.0
```

**Expected behavior**
I would expect instead that a list is passed as configuration

**Configuration (please complete the following information):**
 - OS: MacOS 11.3.1
 - yarp version: YARP version 3.6.0+7-20220107.2+git38581de70

traversaro, 2022-03-09:
Just to give it a shot and to provide more info, can you try also:
~~~
$ yarprobotinterface --config my_application --dIKGains "\(10.0,100.0\)"
~~~
and
~~~
$ yarprobotinterface --config my_application --dIKGains '(10.0,100.0)'
~~~

lrapetti, 2022-03-09:
> Just to give it a shot and to provide more info, can you try also:
> 
> ```
> $ yarprobotinterface --config my_application --dIKGains "\(10.0,100.0\)"
> ```

This gives
```sh
dynamicalIKCorrectionGainsLinRot is (10.0
```

> and
> 
> ```
> $ yarprobotinterface --config my_application --dIKGains '(10.0,100.0)'
> ```

this doesn't change the from what I was getting at the beginning

----- 

I also tried the following
```sh
$ yarprobotinterface --config my_application --dIKGains "\(10.0\,100.0\)"
```
in this case it gets printed correctly:
```sh
dynamicalIKCorrectionGainsLinRot is (10.0,100.0)
```

but then it fails when checking if it is a list, i.e.
```cpp
config.find("dynamicalIKCorrectionGainsLinRot").isList()
```
returns `False`

randaz81, 2022-04-28:
Bug confirmed

pattacini, 2022-05-18:
Hi @randaz81 

Did you see my post above referring to #2836?

randaz81, 2022-05-18:
Yes, I understood that you are in favor of merging that PR which fixes this issue with a simple workaround and does not modifies anything in the current behavior of yarp::os classes. Am I right?

pattacini, 2022-05-19:
Yep, we're aligned 👍🏻 

My only doubt stemmed from the question `should the output of yarp::os::Value.asList()->toString() include the '(' characters?`, which, differently from what I thought, needs to be read within the context of `yarprobotinterface` and not in a general manner 👍🏻

randaz81, 2022-05-19:
fixed by https://github.com/robotology/yarp/pull/2836

closed 2022-05-19 by randaz81

lrapetti, 2022-05-20:
Thanks @randaz81!!

cc @RiccardoGrieco