robotology/yarp #2814: Connection among two ports connected with UDP carrier results in the sender being blocked for 0.25 seconds once the received is closing

traversaro, 2022-03-10:
**Describe the bug**
If one has two BufferedPort connected by and UDP carrier, once the receiver is closed, the sender can be blocked to up to 0.25 seconds. 

**To Reproduce**
A reproducible example can be found in https://gist.github.com/traversaro/d8ed241b0948808620a00c3f74501349 , to compile and run it execute:
~~~
git clone https://gist.github.com/traversaro/d8ed241b0948808620a00c3f74501349
cd d8ed241b0948808620a00c3f74501349
mkdir build
cd build
cmake .
make 
~~~

Then run:
* `yarpserver` 
* `./publish` 
* `launchdriver` 

When you launch `launchdriver` you will see the print (not all the time,  but 2/3 of the times):
~~~
[INFO] Publish duration:  0.247578
~~~

As soon as you change the carrier in https://gist.github.com/traversaro/d8ed241b0948808620a00c3f74501349#file-launchdriver-cpp-L26 to `tcp` or `fast_tcp` and recompile, the problem disappears. 

**Expected behavior**
Even if in general YARP does not make any kind of real time promise, a sudden block of 0.25 seconds is quite surprising.

**Configuration (please complete the following information):**
 - OS: Ubuntu 20.04
 - yarp version: 3.6.0
-  compiler: gcc 9.3.0

**Additional context**
We noticed this because the wholeBodyDynamics was getting stuck for 0.250 seconds when a connection was closed, and this resulted in the torque estimated not be send to the low level firmware for a period of 0.25 seconds, that resulted in the firmware automatically switching to Hardware Fault mode as no torque estimate was received.

traversaro, 2022-03-10:
Knowing that the issue only happens with the `udp` carrier, this left us with a limited amount of classes in which this 0.25 could be:
* https://github.com/robotology/yarp/blob/v3.6.0/src/libYARP_os/src/yarp/os/impl/UdpCarrier.h
* https://github.com/robotology/yarp/blob/v3.6.0/src/libYARP_os/src/yarp/os/impl/UdpCarrier.cpp
* https://github.com/robotology/yarp/blob/v3.6.0/src/libYARP_os/src/yarp/os/impl/DgramTwoWayStream.h
* https://github.com/robotology/yarp/blob/v3.6.0/src/libYARP_os/src/yarp/os/impl/DgramTwoWayStream.cpp

A quick visual inspection revealed what is with all probability the delay that is stopping all the comunications:
* https://github.com/robotology/yarp/blob/702271e6f0fb685847938a2d892c0f629aab29f6/src/libYARP_os/src/yarp/os/impl/DgramTwoWayStream.cpp#L643

Apparently this delay was added in https://github.com/robotology/yarp/commit/e27c89f6daad22ac8290337405ec5e641bae1c7a together with a bunch of multicast-related changes.

traversaro, 2022-03-10:
The fix provided in https://github.com/robotology/yarp/pull/2815 works fine for me, but I am not sure about possible side-effects. I try to analyze why the delay was added in https://github.com/robotology/yarp/commit/e27c89f6daad22ac8290337405ec5e641bae1c7a, but there is no specific reason documented.

randaz81, 2022-03-11:
I'm a bit afraid to change this., even if I do not know the reason for that delay.

When using a UDP carrier, I suggest closing the sender **before** the receiver.
I recommend using another carrier, e.g. fast_tcp, if this is not ok for you.

Indeed, one of the things that people are typically not aware of is that, unlike the tcp protocol, if you kill the receiver while using  UDP, this will keep the OS and the network busy (i.e. it will consume bandwidth), even if no one is receiving data.
This is the way in which udp works by definition, and the main reason why I never use UDP over wifi: it is very easy to lose control of the system. In general, it is a bad idea to have an udp data stream alone. The correct usage of the udp stream (for example in video-meeting applications) is to have a parallel tcp communication to control the flow.

traversaro, 2022-03-11:
> I'm a bit afraid to change this., even if I do not know the reason for that delay.

Ack, I imagined that, I opened https://github.com/robotology/yarp/pull/2815 to at least check if some test was failing if the delay was removed, as unfortunatly I could not find any reason that justified the existence of the delay in the code or in the commit history.

> When using a UDP carrier, I suggest closing the sender before the receiver.

In this specific case this is not possible, as wholeBodyDynanics is not aware of the receivers.

>  I recommend using another carrier, e.g. fast_tcp, if this is not ok for you.

Yes, we will probably do that, thanks for the clarification!

traversaro, 2022-03-11:
@randaz81 at this point I guess we can close this issue or do you think there is any action that is necessary to do?

randaz81, 2022-04-21:
Discussed with DrDanz, and he also recommended keeping the delay.
I'm thus closing the issue.

closed 2022-04-21 by randaz81