robotology/yarp #2817: ROS: Subscriber: Code gets stuck on "subscriber.read(data,false);"

jhielson, 2022-03-16:
Hi all,

I am having some difficulties trying to use ROS and YARP together. I can publish data without a problem. But when I run the command ```subscriber.read(data,false)``` to read one of the topics from my applications that node gets completely stuck there even when I set the "shouldWait" parameter to false. 

Is there a way to create a callback function so my node wouldn't need to wait for a new message? If that is not possible how can it continue the execution of the loop when there is no message to be read?

On the top of my code, I generate a subscriber and publisher:
```
    /* subscribe to cerebellum */
    yarp::os::Subscriber<String> subscriber;
    if (!subscriber.topic("/cerebellumOutput")) {
        yWarning("Failed to subscriber to /cerebellumOutput");
        return -1;
    }

    /* publish to cerebellum */
    yarp::os::Publisher<String> publisher;
    if (!publisher.topic("/cerebellumInputs")) {
        cerr<<"Error opening port, check your yarp network\n";
        return -1;
    } 
```

And, then, I use the command within my loop to read the messages and perform the task:

```
String data;
subscriber.read(data,false);
```

Please, let me know if I have been doing something wrong. 

Best regards,
Jhielson M. Pimentel

randaz81, 2022-04-21:
`read(data, false)` seems a bit strange to me. Can you try something like:

```
yarp::rosmsg::geometry_msgs::PoseWithCovarianceStamped* rospose = m_rosSubscriber_current_pose.read(false);
if (rospose)
{
   //do something here
}
```     

Yes, you can also write a callback, but the code is a bit more complicated. 
You can check `MobileBaseVelocityControl_nws_ros.h` in yarp for an example.

jhielson, 2022-05-03:
Thanks, @randaz81. I believe it is working now. Sorry for the delayed response.

closed 2022-05-23 by jhielson