robotology/yarp #2818: Increase temporal resolution of yarpdataplayer

arrenglover, 2022-03-22:
**Is your feature request related to a problem? Please describe.**

Event-camera datasets have packets with sub-millisecond (typically not less than 0.1 millisecond) timestamps. The yarpdataplayer has a maximum update rate of1ms.  Playing back datasets results in temporal distortion as it cannot play >1 packets in the timeframe. 

**Describe the solution you'd like**

Update the yarpdataplayer to not rely on a base update rate of 1ms. Instead make it asynchronous, (no periodicthread), or at a higher temporal resolution (0.05 ms?).

**Describe alternatives you've considered**

At the moment to play our data back in real-time, we re-process our data to form packets of at-least 4ms. We use 4ms as, 1 ms seems only possible at the 2x speed option, and 4ms gives more tolerance than 2ms. This solution is workable but causes differences between real camera playback and yarpdataplayer playback. It requires any saved data to be re-coded before it can be used.

An alternative to increasing the yarpdataplayer update thread resolution would be to allow more than 1 packet to be sent in the 1ms timeframe.

randaz81, 2022-04-21:
@vtikha do you think it is possibile?

vtikha, 2022-04-21:
We need to try and run some tests. @arrenglover have you tried some of the suggestions yourself? Just to know if you managed to do something.

arrenglover, 2022-04-26:
Hi @vtikha. Unfortunately not sorry - I hope I have understood correctly how the dataplayer is working but I haven't managed to delve into the code with more detail. let me know if you want to meet up to clarify anything. Thanks a lot for looking at this.

arrenglover, 2023-02-22:
I've opened a pull request for this fix #2946