robotology/yarp #2824: Yarpmotorgui target box lands in a different position compared to mouse cursor

mfussi66, 2022-04-13:
**Describe the bug**
When using the sliders in the Yarpmotorgui in position control mode it often happens that the desired (feasible) position target, which is represented by the blue box, is not in the same position as the mouse cursor when releasing the left mouse button.

**To Reproduce**
Instantiated the iCub3 model in gazebo and try moving the head in position mode with the yarpmotorgui

**Expected behavior**
The blue box representing the desired target should end up in the slider position chosen by the mouse cursor.

**Screenshots**
See the video attached. The slider moves automatically in the position set by the mouse cursor, but the blue box translated unreliably.

https://user-images.githubusercontent.com/38140169/163225823-68defed4-07c5-4211-bc57-270206a0cef7.mp4

**Configuration (please complete the following information):**
 - OS: Ubuntu 20.04
 - yarp version: Yarp 3.6.0
-  compiler: gcc 9.4.0
- Qt version: 5.12.8 (provided by default ubuntu repos)

**Additional context**
I tried to take a look at the code and noticed that the position of the box is updated here: https://github.com/robotology/yarp/blob/master/src/yarpmotorgui/sliderWithTarget.cpp#L90-L121=

```cpp
void SliderWithTarget::paintEvent(QPaintEvent *e)
{
    QSlider::paintEvent(e);
    double w = this->width() - 30;
    int totValues = this->maximum() + abs(this->minimum());

    if (enableViewTargetBox)
    {
        QPainter p(this);
        double newX = ((double)w / (double)totValues) * ((double)target + abs(this->minimum()));
        QRect r(newX, 17, 30, 15);
        p.fillRect(r, QBrush(QColor(128, 128, 255, 128)));
        p.drawRect(r);
    }
...
}
```
So maybe `newX` is not properly scaled, unlike the position label, although the factors are the same.

Also, it seems that the farther I move the target away from zero, the farther the box moves.

pattacini, 2022-04-13:
cc @S-Dafarra

elandini84, 2022-05-18:
Dear @mfussi66, this PR should fix this issue: [2842](https://github.com/robotology/yarp/pull/2842)

mfussi66, 2022-05-18:
Thank you @elandini84 , much appreciated!

closed 2022-05-25 by randaz81