robotology/yarp #2839: yarpmotorgui set custom positions of only certain parts of the robot

HosameldinMohamed, 2022-05-02:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.

We use `yarpmotorgui` to set the robot joints to custom positions. But sometimes we are interested in setting the custom positions only to some parts. We update the `.ini` file such that only the desired parts are included. `yarpmotorgui` then throws error messages (see screenshot), with the number of parts not included. However, it still does the function as expected.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

It would be nice to set the custom positions without the error messages.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

[image]

HosameldinMohamed, 2022-05-02:
CC @gabrielenava @FabioBergonti @Giulero @affafjunaidMOMIN @antonellopaolino @traversaro

elandini84, 2022-06-13:
> **Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.
>
> We use `yarpmotorgui` to set the robot joints to custom positions. But [...]

This PR should fix the issue: [2853](https://github.com/robotology/yarp/pull/2853)

closed 2022-06-17 by randaz81

HosameldinMohamed, 2022-06-17:
Thanks @elandini84