robotology/yarp #2863: Missing includes of <utility> ad <iterator>

xEnVrE, 2022-06-27:
**Describe the bug**
I cannot compile `YARP` due to the following errors:

```
/home/xenvre/robot-code/robotology-superbuild/src/YARP/src/libYARP_os/src/yarp/os/Header.cpp: In constructor ‘yarp::os::Header::Header(yarp::os::Header&&)’:
/home/xenvre/robot-code/robotology-superbuild/src/YARP/src/libYARP_os/src/yarp/os/Header.cpp:81:20: error: ‘exchange’ is not a member of ‘std’
   81 |         mPriv(std::exchange(rhs.mPriv, nullptr))
```

```
[  1%] Building CXX object src/libYARP_os/src/CMakeFiles/YARP_os.dir/yarp/os/Log.cpp.o
In file included from /home/xenvre/robot-code/robotology-superbuild/src/YARP/src/libYARP_conf/src/yarp/conf/environment.h:13,
                 from /home/xenvre/robot-code/robotology-superbuild/src/YARP/src/libYARP_os/src/yarp/os/Log.cpp:8:
/home/xenvre/robot-code/robotology-superbuild/src/YARP/src/libYARP_conf/src/yarp/conf/string.h: In function ‘typename ContainerT::value_type yarp::conf::string::join(typename ContainerT::const_iterator, typename ContainerT::const_iterator, const typename ContainerT::value_type&)’:
/home/xenvre/robot-code/robotology-superbuild/src/YARP/src/libYARP_conf/src/yarp/conf/string.h:105:36: error: ‘ostream_iterator’ is not a member of ‘std’
  105 |     std::copy(begin, end - 1, std::ostream_iterator<string_type, char_type, traits_type>(s, separator.c_str()));
```

```
[ 14%] Building CXX object src/libYARP_sig/src/CMakeFiles/YARP_sig.dir/yarp/sig/Image.cpp.o
/home/xenvre/robot-code/robotology-superbuild/src/YARP/src/libYARP_sig/src/yarp/sig/Image.cpp: In constructor ‘yarp::sig::Image::Image(yarp::sig::Image&&)’:
/home/xenvre/robot-code/robotology-superbuild/src/YARP/src/libYARP_sig/src/yarp/sig/Image.cpp:813:27: error: ‘exchange’ is not a member of ‘std’
  813 |     : implementation(std::exchange(other.implementation, nullptr))
      |                           ^~~~~~~~
/home/xenvre/robot-code/robotology-superbuild/src/YARP/src/libYARP_sig/src/yarp/sig/Image.cpp: In member function ‘yarp::sig::Image& yarp::sig::Image::operator=(yarp::sig::Image&&)’:
/home/xenvre/robot-code/robotology-superbuild/src/YARP/src/libYARP_sig/src/yarp/sig/Image.cpp:821:27: error: ‘exchange’ is not a member of ‘std’
  821 |     implementation = std::exchange(other.implementation, nullptr);
      |                           ^~~~~~~~
/home/xenvre/robot-code/robotology-superbuild/src/YARP/src/libYARP_sig/src/yarp/sig/Image.cpp: In member function ‘bool yarp::sig::Image::move(yarp::sig::Image&&)’:
/home/xenvre/robot-code/robotology-superbuild/src/YARP/src/libYARP_sig/src/yarp/sig/Image.cpp:881:31: error: ‘exchange’ is not a member of ‘std’
  881 |         implementation = std::exchange(alt.implementation, nullptr);
      |             
```

which can be solved by adding the following missing includes:
- `#include <utility>` in `src/YARP/src/libYARP_os/src/yarp/os/Header.cpp`
- `#include <iterator>` in `src/YARP/src/libYARP_conf/src/yarp/conf/string.h`
- `#include <utility>` in `src/YARP/src/libYARP_sig/src/yarp/sig/Image.cpp`

**To Reproduce**

I compiled `YARP` using the `robotology-superbuild`.

**Expected behavior**
I expected that `YARP` compiled successfully.

**Configuration (please complete the following information):**
 - OS: `Arch Linux`
 - yarp version: `YARP version 3.7.1+dirty` (https://github.com/robotology/yarp/commits/v3.7.1)
-  compiler: `gcc 12.1.0`

traversaro, 2022-06-28:
Can you provide a PR with the suggested changes? I get it is easier to integrated them in that form. : )

xEnVrE, 2022-06-28:
Done in #2864

randaz81, 2022-06-29:
fixed by https://github.com/robotology/yarp/pull/2864

closed 2022-06-29 by randaz81