robotology/yarp #2865: Unable to compile yarp_3.7

triccyx, 2022-06-29:
**Describe the bug**
Switching from branch yarp-3.6 to yarp-3.7 unable to compile yarp-3.7
The problem seems on somethings about ROS. I have `removed from the compilation anything about ROS`.

**To Reproduce**
Steps to reproduce the behavior:
```
git checkout yarp-3.7
make 
```

**Expected behavior**
YARP compile

**Screenshots**
```
[  1%] Built target YARP_priv_xmlrpcpp
[  1%] Built target YARP_priv_hmac
[ 13%] Built target YARP_os
[ 15%] Built target YARP_sig
[ 15%] Built target YARP_gsl
[ 16%] Built target YARP_math
[ 17%] Built target YARP_init
[ 17%] Built target yarpidl_rosmsg
[ 17%] Generating code from ../../../extern/ros/common_msgs/trajectory_msgs/msg/JointTrajectory.msg
Segmentation fault (core dumped)
make[2]: *** [src/libYARP_rosmsg/src/CMakeFiles/YARP_rosmsg_gen_trajectory_msgs.dir/build.make:92: src/libYARP_rosmsg/src/yarp/rosmsg/trajectory_msgs/JointTrajectory.h] Error 139
make[1]: *** [CMakeFiles/Makefile2:5270: src/libYARP_rosmsg/src/CMakeFiles/YARP_rosmsg_gen_trajectory_msgs.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
```

**Configuration (please complete the following information):**
 - OS: Ubuntu 21.10
 - yarp version: 3.7
-  compiler: GCC 11.2.0

**Additional context**
I don't need ROS, is it possible to prevent
`src/libYARP_rosmsg/src/yarp/rosmsg/trajectory_msgs/JointTrajectory.h`
to be compiled?

randaz81, 2022-06-29:
Have you tried simply a `make clean` or removing your build directory? 
I do not remember any ROS-specific change between yarp 3.6 and 3.7.

traversaro, 2022-06-29:
Are you sure that you do not have an existing YARP installation in your system (with a different version), and perhaps that you are setting `LD_LIBRARY_PATH`? One possible problem is that the `yarpidl_rosmsg` used in the build is linking a wrong YARP version, and this leads to a segfault.

traversaro, 2022-06-29:
Just to have a suggestion for the future, it may be worth to just cancel `build/install`, instead of the complete superbuild. It is possible that the "wrong YARP version around" was in `build/install`.

isorrentino, 2022-06-30:
I had the same error. The @traversaro's suggestion 

> cancel `build/install` 

worked in my case. Thanks a lot @traversaro!

traversaro, 2022-06-30:
Then this problem is actually an instance of the problem described in https://github.com/robotology/robotology-superbuild/issues/97 . Basically `yarpidl_rosmsg` was loading the `libYARP_so.so` (and other libraries) from the `build/install` , instead of the build, due to the presence of `LD_LIBRARY_PATH`  in the setup.sh of the superbuild. Probably we should get rid of that.