robotology/yarp #2872: yarpview segfaults when testing h264

S-Dafarra, 2022-07-18:
**Describe the bug**
I tried to follow https://www.yarp.it/latest//group__carrier__h264__howto.html to test the usage of ``h264`` with ``yarp``. When I try to connect the server to ``yarpview``, ``yarpview`` crashes with ``segfault``.

**To Reproduce**
Run the following on different terminals
### T1
```
gst-launch-1.0 -v videotestsrc ! "video/x-raw, format=(string)I420, width=(int)640, height=(int)480" ! x264enc ! h264parse ! rtph264pay pt=96 config-interval=5 ! udpsink host=127.0.0.1 port=101234
```
### T2
```
yarpserver
```
### T3
```
yarpview
```
###T4
```
yarp register /gst h264 127.0.0.1 101234
yarp connect /gst /yarpview/img:i h264
```

**Expected behavior**
``yarpview`` should not crash. I tried running 
```
gst-launch-1.0 -v udpsrc port=101234 caps="application/x-rtp, media=(string)video, encoding-name=(string)H264, payload=(int)96" ! rtph264depay ! h264parse ! avdec_h264! autovideosink
```
and I can see the image showed in https://www.yarp.it/latest//group__carrier__h264__howto.html

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Configuration (please complete the following information):**
 - OS: Ubuntu 20.04
 - yarp version: 3.7.0
-  compiler: ``gcc``

**Additional context**
Add any other context about the problem here.

S-Dafarra, 2022-07-18:
I checked if it was a problem of ``yarpview`` by using
```
yarpdev --device fakeFrameGrabber --name /grabber
```
but in this case there are no issues.

randaz81, 2023-08-07:
clone of #2289

randaz81, 2024-08-30:
`h264 `carrier replaced by `gstreamer `carrier: https://github.com/robotology/yarp/pull/3127. See updated documentation.

closed 2024-08-30 by randaz81