robotology/yarp #2885: ffmpeg_grabber does not compile against Ffmpeg 5.0

traversaro, 2022-10-06:
Compilation error:
~~~
  yarp_plugin_yarpmod_ffmpeg_grabber.cpp
C:\src\yarp\src\devices\ffmpeg\ffmpeg_api.h(44,25): error C2065: 'AVPicture': undeclared identifier [C:\src\yarp\buildd
ev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\ffmpeg_api.h(44,36): error C2065: 'dst': undeclared identifier [C:\src\yarp\builddev\src
\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\ffmpeg_api.h(44,41): error C2062: type 'int' unexpected [C:\src\yarp\builddev\src\device
s\ffmpeg\yarp_ffmpeg.vcxproj]
  FfmpegGrabber.cpp
C:\src\yarp\src\devices\ffmpeg\ffmpeg_api.h(44,25): error C2065: 'AVPicture': undeclared identifier [C:\src\yarp\buildd
ev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\ffmpeg_api.h(44,36): error C2065: 'dst': undeclared identifier [C:\src\yarp\builddev\src
\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\ffmpeg_api.h(44,41): error C2062: type 'int' unexpected [C:\src\yarp\builddev\src\device
s\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(121,47): error C2039: 'codec': is not a member of 'AVStream' [C:\src\y
arp\builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:/Users/STraversaro/AppData/Local/mambaforge/envs/yarpdev/Library/include\libavformat/avformat.h(950): message : see d
eclaration of 'AVStream' [C:\src\yarp\builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(124,1): error C2440: '=': cannot convert from 'const AVCodec *' to 'AV
Codec *' [C:\src\yarp\builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(124,38): message : Conversion loses qualifiers [C:\src\yarp\builddev\s
rc\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(153,22): error C3861: 'avpicture_get_size': identifier not found [C:\s
rc\yarp\builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(158,25): error C2065: 'AVPicture': undeclared identifier [C:\src\yarp\
builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(158,36): error C2059: syntax error: ')' [C:\src\yarp\builddev\src\devi
ces\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(215,21): error C3861: 'avcodec_decode_audio4': identifier not found [C
:\src\yarp\builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(254,9): error C3861: 'avcodec_decode_video2': identifier not found [C:
\src\yarp\builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(273,46): error C2065: 'AVPicture': undeclared identifier [C:\src\yarp\
builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(273,56): error C2059: syntax error: ')' [C:\src\yarp\builddev\src\devi
ces\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(408,81): error C2440: '=': cannot convert from 'const AVInputFormat *'
 to 'AVInputFormat *' [C:\src\yarp\builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(408,35): message : Conversion loses qualifiers [C:\src\yarp\builddev\s
rc\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(441,45): error C2440: '=': cannot convert from 'const AVInputFormat *'
 to 'AVInputFormat *' [C:\src\yarp\builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(441,35): message : Conversion loses qualifiers [C:\src\yarp\builddev\s
rc\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(490,5): error C3861: 'av_register_all': identifier not found [C:\src\y
arp\builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegGrabber.cpp(701,13): error C3861: 'av_free_packet': identifier not found [C:\src\y
arp\builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
  FfmpegWriter.cpp
C:\src\yarp\src\devices\ffmpeg\ffmpeg_api.h(44,25): error C2065: 'AVPicture': undeclared identifier [C:\src\yarp\buildd
ev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\ffmpeg_api.h(44,36): error C2065: 'dst': undeclared identifier [C:\src\yarp\builddev\src
\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\ffmpeg_api.h(44,41): error C2062: type 'int' unexpected [C:\src\yarp\builddev\src\device
s\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegWriter.cpp(95,13): error C2039: 'codec': is not a member of 'AVStream' [C:\src\yar
p\builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:/Users/STraversaro/AppData/Local/mambaforge/envs/yarpdev/Library/include\libavformat/avformat.h(950): message : see d
eclaration of 'AVStream' [C:\src\yarp\builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegWriter.cpp(112,13): error C2039: 'codec': is not a member of 'AVStream' [C:\src\ya
rp\builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:/Users/STraversaro/AppData/Local/mambaforge/envs/yarpdev/Library/include\libavformat/avformat.h(950): message : see d
eclaration of 'AVStream' [C:\src\yarp\builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegWriter.cpp(115,46): error C2440: '=': cannot convert from 'const AVCodec *' to 'AV
Codec *' [C:\src\yarp\builddev\src\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegWriter.cpp(115,33): message : Conversion loses qualifiers [C:\src\yarp\builddev\sr
c\devices\ffmpeg\yarp_ffmpeg.vcxproj]
C:\src\yarp\src\devices\ffmpeg\FfmpegWriter.cpp(138,20): error C2039: 'codec': is not a member of 'AVStream' [C:\src\ya
[... 194 lines cut ...]
2022-10-06T13:45:03.1464283Z       |                        ^
2022-10-06T13:45:03.1464626Z In file included from /usr/share/miniconda3/envs/test/include/libavcodec/bsf.h:30,
2022-10-06T13:45:03.1465072Z                  from /usr/share/miniconda3/envs/test/include/libavcodec/avcodec.h:44,
2022-10-06T13:45:03.1465759Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/YARP/src/devices/ffmpeg/FfmpegWriter.h:13,
2022-10-06T13:45:03.1466539Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/YARP/src/devices/ffmpeg/FfmpegWriter.cpp:38:
2022-10-06T13:45:03.1467062Z /usr/share/miniconda3/envs/test/include/libavcodec/packet.h:488:6: note: declared here
2022-10-06T13:45:03.1467455Z   488 | void av_init_packet(AVPacket *pkt);
2022-10-06T13:45:03.1467735Z       |      ^~~~~~~~~~~~~~
2022-10-06T13:45:03.1631108Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/YARP/src/devices/ffmpeg/FfmpegWriter.cpp:216:13: warning: 'AVStream::codec' is deprecated [-Wdeprecated-declarations]
2022-10-06T13:45:03.1632318Z   216 |     c = st->codec;
2022-10-06T13:45:03.1632571Z       |             ^~~~~
2022-10-06T13:45:03.1633192Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/src/YARP/src/devices/ffmpeg/FfmpegWriter.h:14,
2022-10-06T13:45:03.1634012Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/YARP/src/devices/ffmpeg/FfmpegWriter.cpp:38:
2022-10-06T13:45:03.1634565Z /usr/share/miniconda3/envs/test/include/libavformat/avformat.h:888:21: note: declared here
2022-10-06T13:45:03.1635328Z   888 |     AVCodecContext *codec;
2022-10-06T13:45:03.1635590Z       |                     ^~~~~
2022-10-06T13:45:03.1636538Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/YARP/src/devices/ffmpeg/FfmpegWriter.cpp:216:13: warning: 'AVStream::codec' is deprecated [-Wdeprecated-declarations]
2022-10-06T13:45:03.1637141Z   216 |     c = st->codec;
2022-10-06T13:45:03.1637384Z       |             ^~~~~
2022-10-06T13:45:03.1637985Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/src/YARP/src/devices/ffmpeg/FfmpegWriter.h:14,
2022-10-06T13:45:03.1638791Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/YARP/src/devices/ffmpeg/FfmpegWriter.cpp:38:
2022-10-06T13:45:03.1639341Z /usr/share/miniconda3/envs/test/include/libavformat/avformat.h:888:21: note: declared here
2022-10-06T13:45:03.1639703Z   888 |     AVCodecContext *codec;
2022-10-06T13:45:03.1639968Z       |                     ^~~~~
~~~

traversaro, 2022-10-06:
PR that fixes a similar problem in Gazebo Classic.

traversaro, 2023-07-11:
It interesting to note that the `image_compression_ffmpeg` portmonitor instead compiles fine with both ffmpeg 5 and 6, so that one can compile fine with recent ffmpeg.

EhsanRanjbari, 2024-01-04:
I do not know if it is related to this issue or not but I am having a compilation error as below when the option `ENABLE_yarpcar_portmonitor` is `ON`: (https://github.com/ami-iit/element_haptic-gloves/issues/159#issuecomment-1805295330)

```bash
[ 50%] Building CXX object src/portmonitors/image_compression_ffmpeg/CMakeFiles/yarp_pm_image_compression_ffmpeg.dir/ffmpegPortmonitor.cpp.o
[ 50%] Linking CXX shared module ../../../lib/yarp/yarp_pm_image_compression_ffmpeg.so
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libswscale.a(hscale_fast_bilinear_simd.o): relocation R_X86_64_32S against `.text.unlikely' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: failed to set dynamic section sizes: bad value
collect2: error: ld returned 1 exit status
make[5]: *** [src/portmonitors/image_compression_ffmpeg/CMakeFiles/yarp_pm_image_compression_ffmpeg.dir/build.make:118: lib/yarp/yarp_pm_image_compression_ffmpeg.so] Error 1
make[4]: *** [CMakeFiles/Makefile2:5623: src/portmonitors/image_compression_ffmpeg/CMakeFiles/yarp_pm_image_compression_ffmpeg.dir/all] Error 2
make[3]: *** [Makefile:156: all] Error 2
make[2]: *** [CMakeFiles/YARP.dir/build.make:85: src/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:1083: CMakeFiles/YARP.dir/all] Error 2
make: *** [Makefile:101: all] Error 2
```

traversaro, 2024-01-04:
No, that error is different. Can you open a new issue describing the environment in which you are experiencing this problem?

randaz81, 2024-09-16:
We decided to deprecated device ffmpeg_grabber  and ffmpeg_writer.
The compilation of these plugins has been disabled and they will be removed in the near future.

For a yarp device with compatible functionalities, please use opencv_grabber

closed 2024-09-16 by randaz81