robotology/yarp #2892: `IMURosPublisher` publishes rpy data in wrong format

Nicogene, 2022-11-14:
**Describe the bug**
The imu msg represents the orientation as a quaternion (see [here](http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/Imu.html)) `IMURosPublisher` publishes the data  seems to publish it as rpy:

https://github.com/robotology/yarp/blob/f03dd5d71a78db86486dd006d8855af993874b9a/src/devices/multipleAnalogSensorsRosPublishers/IMURosPublisher.cpp#L64-L66

**To Reproduce**

**Expected behavior**
The orientation is published as quaternion

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Configuration (please complete the following information):**
 - OS: WSL2 - Windows 11
 - yarp version: `3.7.2+1-20220630.8+gitc8450ac9e`
-  compiler:

**Additional context**
Add any other context about the problem here.

cc @traversaro @randaz81 @elandini84

traversaro, 2022-11-14:
Another related piece of code: https://github.com/robotology/yarp/blob/f03dd5d71a78db86486dd006d8855af993874b9a/src/devices/multipleAnalogSensorsRosPublishers/PoseStampedRosPublisher.cpp#L53 .

randaz81, 2022-11-14:
Fixed as suggested by @traversaro

closed 2022-11-14 by randaz81