robotology/yarp #2896: Cannot run the yarpserver from the Cluster Management Window of yarpmanager

xEnVrE, 2022-11-21:
**Describe the bug**
I cannot run the `yarpserver` from the `Cluster Management Window` of the `yarpmanager`.

[image]

Whenever I try to press the green button, I get the following in the terminal when I run the `yarpmanager`:

```console
[ERROR] ClusterWidget: yarpserver is not running
```

**To Reproduce**

Run the `yarpmanager` and try to run the `yarpserver` from the  `Cluster Management Window` of the `yarpmanager`.

**Expected behavior**

The `yarpserver` starts after pressing on the green button of the  `Cluster Management Window` of the `yarpmanager`.

**Configuration (please complete the following information):**
 - OS: `Ubuntu 20.04` (running inside Docker)
 - yarp version: `YARP version 3.7.2`
-  compiler: `g++ (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0`

**Additional context**

I am running the latest distro that has been installed using the `robotology-superbuild`. `git status` executed within the superbuild folder reads
```console
HEAD detached at v2022.09.1
```

xEnVrE, 2022-11-21:
I also tested with the latest yarp `YARP version 3.7.2+92-20221116.5+git90cbb8697` and I got the same result.

randaz81, 2022-11-21:
Not sure about this issue, it could be also setup-dependent.
Can you check if you ssh iiticublap202 and verify also if xenvre is a valid username (and its permissions)?

xEnVrE, 2022-11-21:
Hi @randaz81, it was a connectivity issue, such that the ssh connection was not working properly.

Closing then.

closed 2022-11-21 by xEnVrE