robotology/yarp #2909: macOS configuration with CMake 3.25.1 fails

traversaro, 2022-12-01:
**Describe the bug**
Configuration fails with:
~~~
2022-12-01T19:35:05.2691860Z CMake Error at cmake/YarpMacOSUtilities.cmake:106 (set_target_properties):
2022-12-01T19:35:05.2692950Z   CXX_MODULE_HEADER_UNIT_SETS property is read-only
2022-12-01T19:35:05.2693420Z 
2022-12-01T19:35:05.2693900Z Call Stack (most recent call first):
2022-12-01T19:35:05.2694600Z   src/yarpview/src/CMakeLists.txt:72 (yarp_macos_duplicate_and_add_bundle)
2022-12-01T19:35:05.2695010Z 
2022-12-01T19:35:05.2695270Z 
2022-12-01T19:35:05.2695830Z CMake Error at cmake/YarpMacOSUtilities.cmake:106 (set_target_properties):
2022-12-01T19:35:05.2696890Z   CXX_MODULE_SETS property is read-only
2022-12-01T19:35:05.2697260Z 
2022-12-01T19:35:05.2697730Z Call Stack (most recent call first):
2022-12-01T19:35:05.2698400Z   src/yarpview/src/CMakeLists.txt:72 (yarp_macos_duplicate_and_add_bundle)
2022-12-01T19:35:05.2698810Z 
2022-12-01T19:35:05.2699070Z 
2022-12-01T19:35:05.2699630Z CMake Error at cmake/YarpMacOSUtilities.cmake:106 (set_target_properties):
2022-12-01T19:35:05.2700780Z   INTERFACE_CXX_MODULE_HEADER_UNIT_SETS property is read-only
2022-12-01T19:35:05.2701210Z 
2022-12-01T19:35:05.2701640Z Call Stack (most recent call first):
2022-12-01T19:35:05.2702270Z   src/yarpview/src/CMakeLists.txt:72 (yarp_macos_duplicate_and_add_bundle)
2022-12-01T19:35:05.2702760Z 
2022-12-01T19:35:05.2702960Z 
2022-12-01T19:35:05.2703510Z CMake Error at cmake/YarpMacOSUtilities.cmake:106 (set_target_properties):
2022-12-01T19:35:05.2704690Z   INTERFACE_CXX_MODULE_SETS property is read-only
2022-12-01T19:35:05.2705100Z 
2022-12-01T19:35:05.2705600Z Call Stack (most recent call first):
2022-12-01T19:35:05.2706250Z   src/yarpview/src/CMakeLists.txt:72 (yarp_macos_duplicate_and_add_bundle)
2022-12-01T19:35:05.2706680Z 
2022-12-01T19:35:05.2706920Z 
2022-12-01T19:35:05.2707470Z CMake Error at cmake/YarpMacOSUtilities.cmake:106 (set_target_properties):
2022-12-01T19:35:05.2708570Z   CXX_MODULE_HEADER_UNIT_SETS property is read-only
2022-12-01T19:35:05.2708970Z 
2022-12-01T19:35:05.2709460Z Call Stack (most recent call first):
2022-12-01T19:35:05.2710100Z   src/yarpscope/src/CMakeLists.txt:75 (yarp_macos_duplicate_and_add_bundle)
2022-12-01T19:35:05.2710530Z 
2022-12-01T19:35:05.2710770Z 
2022-12-01T19:35:05.2711340Z CMake Error at cmake/YarpMacOSUtilities.cmake:106 (set_target_properties):
2022-12-01T19:35:05.2712380Z   CXX_MODULE_SETS property is read-only
2022-12-01T19:35:05.2712810Z 
2022-12-01T19:35:05.2713300Z Call Stack (most recent call first):
2022-12-01T19:35:05.2713950Z   src/yarpscope/src/CMakeLists.txt:75 (yarp_macos_duplicate_and_add_bundle)
2022-12-01T19:35:05.2714370Z 
2022-12-01T19:35:05.2714610Z 
2022-12-01T19:35:05.2715180Z CMake Error at cmake/YarpMacOSUtilities.cmake:106 (set_target_properties):
2022-12-01T19:35:05.2716290Z   INTERFACE_CXX_MODULE_HEADER_UNIT_SETS property is read-only
2022-12-01T19:35:05.2716730Z 
2022-12-01T19:35:05.2717210Z Call Stack (most recent call first):
2022-12-01T19:35:05.2717870Z   src/yarpscope/src/CMakeLists.txt:75 (yarp_macos_duplicate_and_add_bundle)
2022-12-01T19:35:05.2718290Z 
2022-12-01T19:35:05.2718520Z 
2022-12-01T19:35:05.2719100Z CMake Error at cmake/YarpMacOSUtilities.cmake:106 (set_target_properties):
2022-12-01T19:35:05.2720160Z   INTERFACE_CXX_MODULE_SETS property is read-only
2022-12-01T19:35:05.2720570Z 
2022-12-01T19:35:05.2721320Z Call Stack (most recent call first):
2022-12-01T19:35:05.2721990Z   src/yarpscope/src/CMakeLists.txt:75 (yarp_macos_duplicate_and_add_bundle)
2022-12-01T19:35:05.2722400Z 
2022-12-01T19:35:05.2722640Z 
2022-12-01T19:35:05.2723220Z CMake Error at cmake/YarpMacOSUtilities.cmake:106 (set_target_properties):
2022-12-01T19:35:05.2724320Z   CXX_MODULE_HEADER_UNIT_SETS property is read-only
2022-12-01T19:35:05.2724730Z 
2022-12-01T19:35:05.2725140Z Call Stack (most recent call first):
2022-12-01T19:35:05.2725790Z   src/yarpmanager/CMakeLists.txt:195 (yarp_macos_duplicate_and_add_bundle)
2022-12-01T19:35:05.2726210Z 
2022-12-01T19:35:05.2726700Z 
2022-12-01T19:35:05.2727260Z CMake Error at cmake/YarpMacOSUtilities.cmake:106 (set_target_properties):
2022-12-01T19:35:05.2728340Z   CXX_MODULE_SETS property is read-only
2022-12-01T19:35:05.2728720Z 
2022-12-01T19:35:05.2729200Z Call Stack (most recent call first):
[... 167 lines cut ...]
2022-12-01T19:35:05.8385100Z Call Stack (most recent call first):
2022-12-01T19:35:05.8386070Z   src/yarplaserscannergui/CMakeLists.txt:72 (yarp_macos_duplicate_and_add_bundle)
2022-12-01T19:35:05.8387090Z 
2022-12-01T19:35:05.8387330Z 
2022-12-01T19:35:05.8388880Z CMake Error at cmake/YarpMacOSUtilities.cmake:106 (set_target_properties):
2022-12-01T19:35:05.8391240Z   INTERFACE_CXX_MODULE_HEADER_UNIT_SETS property is read-only
2022-12-01T19:35:05.8392300Z 
2022-12-01T19:35:05.8392800Z Call Stack (most recent call first):
2022-12-01T19:35:05.8393400Z   src/yarplaserscannergui/CMakeLists.txt:72 (yarp_macos_duplicate_and_add_bundle)
2022-12-01T19:35:05.8393830Z 
2022-12-01T19:35:05.8394410Z 
2022-12-01T19:35:05.8396100Z CMake Error at cmake/YarpMacOSUtilities.cmake:106 (set_target_properties):
2022-12-01T19:35:05.8397940Z   INTERFACE_CXX_MODULE_SETS property is read-only
2022-12-01T19:35:05.8398880Z 
2022-12-01T19:35:05.8400040Z Call Stack (most recent call first):
2022-12-01T19:35:05.8404700Z   src/yarplaserscannergui/CMakeLists.txt:72 (yarp_macos_duplicate_and_add_bundle)
2022-12-01T19:35:05.8405790Z 
~~~

**To Reproduce**
Steps to reproduce the behavior:
* Configure latest (as of 2022/12/1) build of YARP with CMake 3.25.1 
 
**Expected behavior**
Configuration is successful.

traversaro, 2022-12-01:
This is similar to https://github.com/robotology/robotology-superbuild/issues/1090, that was fixed in YARP by https://github.com/robotology/yarp/pull/2822 and https://github.com/robotology/yarp/pull/1635 . In this case I do not know how to fix the issue as far as I know there is no programmatic way to check if a property is read-only in CMake, so we can just disable bundles on macOS as we did in the superbuild in https://github.com/robotology/robotology-superbuild/pull/1092, by setting `YARP_DISABLE_MACOS_BUNDLES` option to `ON`. 

All the bundle machinery was added back in time by @francesco-romano (see https://github.com/robotology/yarp/pull/606), and people interested in these feature were macOS that discussed in https://github.com/robotology/yarp/issues/580, such as @kt10aan @vtikha @claudiofantacci . However, unless there is some macOS user interested in this feature, probably it make more sense to drop it (i.e. remove all code not executed if `YARP_DISABLE_MACOS_BUNDLES` is `ON`) to reduce the mantainance burden.

traversaro, 2022-12-01:
> However, unless there is some macOS user interested in this feature, probably it make more sense to drop it (i.e. remove all code not executed if `YARP_DISABLE_MACOS_BUNDLES` is `ON`) to reduce the mantainance burden.

As a reference, I disabled macOS bundles in the superbuild back in April (see https://github.com/robotology/yarp/issues/2909) and nobody complained as far as I know.

kt10aan, 2022-12-02:
I still use YARP on macOS, I will try to give the issue a go, but it will take a few days as I am upgrading to Ventura now.

traversaro, 2022-12-02:
Great, thanks a lot @kt10aan ! Based on my initial super-local assessment the issue could be fixed if we had a way to understand which CMake properties are read-only, but perhaps there is some better solution that avoids to set/get some many arbitrary properties at all.

kt10aan, 2022-12-23:
Sorry for getting late with that. Apparently, as far as I can tell, the. quickest (but not that quick) solution would be some loop to get/set for all properties (maybe fewer with some regular expressions)  or just hardcode them. Maybe this will change in some future version of CMake.

Btw, I have stopped using the bundle option a long time ago.
Tested with Ventura and CMake 3.25.1