robotology/yarp #2914: yarpmotogui - add Select/Deselect all parts Button

davidetome, 2022-12-19:
It could be nice to have a button to select/deselect all parts of the robot in the `yarpmotorgui`

@randaz81 , I can work on this today (https://github.com/icub-tech-iit/fix/issues/1283) and open a PR when done.

cc @pattacini

davidetome, 2022-12-20:
Ciao @randaz81 ,

 I've opened a PR for this : 

- https://github.com/robotology/yarp/pull/2915

cc @pattacini

pattacini, 2022-12-21:
Hi @davidetome 

A couple of points:
- Probably, the CI complains about these [empty lines](https://github.com/robotology/yarp/pull/2915/files#diff-6be8b80defb22a8c45022fde9535305d1c65ff748365d6ab40982df8497de32cR12-R13).
- We would need to address this [comment](https://github.com/robotology/yarp/pull/2915#issuecomment-1359462838) as well.

closed 2023-01-31 by randaz81