robotology/yarp #2925: Status of YARP CI (updated 14/01/2023)

randaz81, 2023-01-14:
On 14/01/2023, after the migration to Catch3 and YCM 0.15.1 (https://github.com/robotology/yarp/pull/2924), I am reporting the following errors on CI (still to be fixed)

[[21][ubuntu-20.04|gcc|Unix Makefiles ( )|Debug] Valgrind (bindings, failing tests and examples as tests disabled)]
(https://github.com/robotology/yarp/actions/runs/3918198395/jobs/6698850228#logs)
- 	 20 - dev::RGBDSensorClientTest (Timeout)
-	103 - serversql::ServerTest (Failed)

[[22][ubuntu-20.04|gcc|Unix Makefiles ( )|Debug] Valgrind, No ACE (bindings, failing tests and examples as tests disabled)](https://github.com/robotology/yarp/actions/runs/3917874996/jobs/6697946648#logs)	
-	 17 - dev::Rangefinder2DClientTest (Failed)
-	 20 - dev::RGBDSensorClientTest (Timeout)
-	103 - serversql::ServerTest (Failed)

[Codacy - coverage]
(https://github.com/robotology/yarp/actions/runs/3918198395/jobs/6698849219#logs)
-	139 - idl::thrift::demo::run (Failed)
-	147 - idl::rosmsg::demo::run (Failed)
	
[SonarCloud]
(https://github.com/robotology/yarp/actions/runs/3918198395/jobs/6698849298#logs)
-	124 - idl::thrift::demo::run (Failed)
-	132 - idl::rosmsg::demo::run (Failed)
	
The last two errors are due to:
```
124: /usr/bin/ld: /home/runner/work/yarp/yarp/build/lib/libYARP_catch2.a(catch_amalgamated.cpp.o): in function `_sub_I_00100_0':
124: /home/runner/work/yarp/yarp/extern/catch2/catch2/catch_amalgamated.cpp:10512: undefined reference to `__gcov_init'
124: /usr/bin/ld: /home/runner/work/yarp/yarp/build/lib/libYARP_catch2.a(catch_amalgamated.cpp.o): in function `_sub_D_00100_1':
124: /home/runner/work/yarp/yarp/extern/catch2/catch2/catch_amalgamated.cpp:10512: undefined reference to `__gcov_exit'
124: /usr/bin/ld: /home/runner/work/yarp/yarp/build/lib/libYARP_catch2.a(catch_amalgamated.cpp.o):(.data.rel+0x20): undefined reference to `__gcov_merge_add'
124: collect2: error: ld returned 1 exit status
124: make[2]: *** [CMakeFiles/demo_yarp_add_idl.dir/build.make:590: demo_yarp_add_idl] Error 1
124: make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/demo_yarp_add_idl.dir/all] Error 2
124: make: *** [Makefile:95: all] Error 2
```

randaz81, 2023-01-19:
In addition to these issue, PR #2920 introduces the file https://github.com/robotology/yarp/blob/master/cmake/YarpCatchUtils.cmake
The scope of the two variables ${YARP_TEST_LAUNCHER} and ${YARP_TEST_TIMEOUT} needs to be checked.

fbrand-new, 2023-09-04:
Codacy and Sonar should have been solved by https://github.com/robotology/yarp/pull/3004

closed 2025-01-22 by randaz81