robotology/yarp #2955: yarpmotorgui using 80% of the CPU

xEnVrE, 2023-03-09:
**Describe the bug**
Today I noticed that the laptop fan was spinning quite a lot and the only substantial executable running was the `yarpmotorgui`.

[image]

From the screen it seems that the `yarpmotorgui` is using almost 90% of the CPU.

**To Reproduce**
Open the `yarpmotorgui` and wait a bit.

**Expected behavior**
That the `yarpmotorgui` do not consume that amount of CPU.

**Screenshots**
See above.

**Configuration**
 - OS: Ubuntu 22.04
 - yarp version: 
    - ```console
      commit 81917f6d0d980fc9d4ad396c5fb07f7908cfb7f9             
      Author: robotology-bot (Tag Repository Action) <robotology@iit.it>              
      Date:   Mon Mar 6 10:39:18 2023 +0000
              YARP 3.8.0
      ```                                         
cc @randaz81 @traversaro

traversaro, 2023-03-09:
Where you doing something particular with the `yarpmotorgui` or it was just open?

xEnVrE, 2023-03-09:
It was just open.

traversaro, 2023-03-09:
If you are able to reproduce, it may be worth to collect some trace, for example via magic-trace ( https://github.com/janestreet/magic-trace, it should be sufficient to run:
~~~
magic-trace run `which yarpmotorgui`
~~~
 and upload (or inspect yourself) the resulting `trace.fxt` file.

xEnVrE, 2023-03-09:
I will do it tomorrow, it was happening on a plain Linux machine.

traversaro, 2023-03-09:
Note that I always used magic-trace attach-pid, but `magic-trace run` should also work fine, but if something is unexpcted you can also use `magic-trace attach-pid`.

PeterBowman, 2023-03-10:
Hello, I can reproduce this on YARP version 3.8.0+3-20230306.4+git282a223a7 and Ubuntu 20.04. Here is my trace: [trace.zip](https://github.com/robotology/yarp/files/10941504/trace.zip). I used the `run` command, but previously I had to set `/proc/sys/kernel/perf_event_paranoid` to `-1` (was `4`).

> (...) it should be sufficient to run:
> ```
> magic-trace run `which yarpmotorgui`

There is a window prompt that shows up right after launching yarpmotorgui that allows me to select which parts I want to control. CPU usage is negligible in this step, but it rockets up to 100% once the main window is opened.