robotology/yarp #2956: Decrease the minimum depth of the depthimage_to_rgb portmonitor

xEnVrE, 2023-03-14:
**Is your feature request related to a problem? Please describe.**
We started using a new RealSense, namely the short-range camera D405, which can provide good depth maps for object as near as 7 cm to the camera.

As we tipically do, it is always useful to visualize the depth map in a `yarpview` using the `depthimage_to_rgb` portmonitor for debugging or just to quickly check that the camera is working as expected.

Unfortunately, the `depthimage_to_rgb` portmonitor is ignoring any value of depth lower than `0.2 m`:

https://github.com/robotology/yarp/blob/51292e3a53ebb6a1198913d41ce54d25082accb7/src/portmonitors/depthimage_to_rgb/DepthImage2.cpp#L52

**Describe the solution you'd like**
Would it be possible to lower that value to something smaller, e.g. `0.01 m`?

I can open a PR in case.

**Additional context**

This is an example of the depth visualization using the current configuration, `min = 0.2`:

[image]

while this is the same scene after changing `min` to `0.01`:

[image]

cc @randaz81

xEnVrE, 2023-03-15:
After talking with @Nicogene, we realized that the minimum is limited to `0.2` also for the `depthimage_to_mono` portmonitor:

https://github.com/robotology/yarp/blob/51292e3a53ebb6a1198913d41ce54d25082accb7/src/portmonitors/depthimage_to_mono/DepthImage.cpp#L28

randaz81, 2023-03-20:
I think the best way to proceed is to add a run-time parameter to allow users to choose max and min

randaz81, 2023-03-20:
Even if is a new feature, I think we can handle this as a bugfix. I'll thus proceed on the yarp 3.8 branch.

xEnVrE, 2023-09-22:
cc @Arya07 @pestrstr