robotology/yarp #2971: yarpmotorgui does not save sequences

isorrentino, 2023-05-18:
**Describe the bug**
I cannot save a sequence from the `yarpmotorgui`. The GUI for the saving does not appear and the `yarpmotorgui` application remains freezed.

**To Reproduce**
Here is the video showing the steps:

https://github.com/robotology/yarp/assets/43743081/07b8d8ad-bc33-438c-a71b-fe5f0ee67ba2

**Configuration (please complete the following information):**
 - OS: Ubuntu
 - yarp version: yarp-3.8

traversaro, 2023-05-18:
I just tested on WSL2 with yarp/gazebo-yarp-plugins/icub-models installed from conda-forge binaries, and it is working fine, with this env:
~~~
(robsub) traversaro@IITICUBLAP257:~$ mamba list
List of packages in environment: "/home/traversaro/mambaforge/envs/robsub"

  Name                       Version       Build                Channel
─────────────────────────────────────────────────────────────────────────────
  _libgcc_mutex              0.1           conda_forge          conda-forge
  _openmp_mutex              4.5           2_gnu                conda-forge
  ace                        7.1.0         hcb278e6_0           conda-forge
  alsa-lib                   1.2.8         h166bdaf_0           conda-forge
  aom                        3.5.0         h27087fc_0           conda-forge
  assimp                     5.2.5         hf40c2ba_0           conda-forge
  atk-1.0                    2.38.0        hd4edc92_1           conda-forge
  attr                       2.5.1         h166bdaf_1           conda-forge
  blosc                      1.21.4        h0f2a231_0           conda-forge
  boost-cpp                  1.78.0        h6582d0a_3           conda-forge
  bullet-cpp                 3.21          h8b32b4d_4           conda-forge
  bzip2                      1.0.8         h7f98852_4           conda-forge
  c-ares                     1.19.0        hd590300_0           conda-forge
  ca-certificates            2023.5.7      hbcca054_0           conda-forge
  cairo                      1.16.0        h35add3b_1015        conda-forge
  cfitsio                    4.2.0         hd9d235c_0           conda-forge
  console_bridge             1.0.2         h924138e_1           conda-forge
  cppzmq                     4.9.0         he78764c_0           conda-forge
  curl                       8.1.0         h409715c_0           conda-forge
  dartsim                    6.13.0        hbbc9fda_0           conda-forge
  dav1d                      1.2.0         hd590300_0           conda-forge
  dbus                       1.13.6        h5008d03_3           conda-forge
  eigen                      3.4.0         h4bd325d_0           conda-forge
  expat                      2.5.0         hcb278e6_1           conda-forge
  fcl                        0.7.0         hdebe893_2           conda-forge
  ffmpeg                     5.1.2         gpl_h9866456_109     conda-forge
  flann                      1.9.1         h0224099_1012        conda-forge
  fmt                        9.1.0         h924138e_0           conda-forge
  font-ttf-dejavu-sans-mono  2.37          hab24e00_0           conda-forge
  font-ttf-inconsolata       3.000         h77eed37_0           conda-forge
  font-ttf-source-code-pro   2.038         h77eed37_0           conda-forge
  font-ttf-ubuntu            0.83          hab24e00_0           conda-forge
  fontconfig                 2.14.2        h14ed4e7_0           conda-forge
  fonts-conda-ecosystem      1             0                    conda-forge
  fonts-conda-forge          1             0                    conda-forge
  freeglut                   3.2.2         h9c3ff4c_1           conda-forge
  freeimage                  3.18.0        hac4c017_14          conda-forge
  freetype                   2.12.1        hca18f0e_1           conda-forge
  freexl                     1.0.6         h166bdaf_1           conda-forge
  fribidi                    1.0.10        h36c2ea0_0           conda-forge
  gazebo                     11.12.0       hfd3773a_12          conda-forge
  gazebo-yarp-plugins        4.6.0         ha770c72_3           conda-forge
  gdbm                       1.18          h0a1914f_2           conda-forge
  gdk-pixbuf                 2.42.10       h6b639ba_2           conda-forge
  geos                       3.11.2        hcb278e6_0           conda-forge
  geotiff                    1.7.1         h480ec47_8           conda-forge
  gettext                    0.21.1        h27087fc_0           conda-forge
  giflib                     5.2.1         h0b41bf4_3           conda-forge
  glib                       2.76.2        hfc55251_0           conda-forge
  glib-tools                 2.76.2        hfc55251_0           conda-forge
  gmp                        6.2.1         h58526e2_0           conda-forge
  gnutls                     3.7.8         hf3e180e_0           conda-forge
  graphite2                  1.3.13        h58526e2_1001        conda-forge
  graphviz                   8.0.5         h28d9a01_0           conda-forge
  gst-plugins-base           1.22.0        h4243ec0_2           conda-forge
  gstreamer                  1.22.0        h25f0c4b_2           conda-forge
  gtk2                       2.24.33       h90689f9_2           conda-forge
  gts                        0.7.6         h64030ff_2           conda-forge
  harfbuzz                   7.3.0         hdb3a94d_0           conda-forge
  hdf4                       4.2.15        h501b40f_6           conda-forge
  hdf5                       1.14.0        nompi_hb72d44e_103   conda-forge
  icu                        72.1          hcb278e6_0           conda-forge
  icub-models                2.0.1         py311hcafe171_0      conda-forge
[... 173 lines cut ...]
  xorg-libxau                1.0.9         h7f98852_0           conda-forge
  xorg-libxaw                1.0.14        h7f98852_1           conda-forge
  xorg-libxdmcp              1.1.3         h7f98852_0           conda-forge
[... cut ...]
  yaml                       0.2.5         h7f98852_2           conda-forge
  yarp-cxx                   3.8.0         h5ee7595_2           conda-forge
  ycm-cmake-modules          0.15.2        h59595ed_0           conda-forge
  zeromq                     4.3.4         h9c3ff4c_1           conda-forge
  zlib                       1.2.13        h166bdaf_4           conda-forge
  zstd                       1.5.2         h3eb15da_6           conda-forge
  zziplib                    0.13.69       h27826a3_1           conda-forge
~~~

Which Ubuntu version are you using?

isorrentino, 2023-05-18:
So maybe the error is caused by some broken package on my laptop? I'm using Ubuntu 20.04.

closed 2025-03-23 by randaz81