robotology/yarp #3036: cannot launch the wholebodydynamics and gravitycompensater when doing the redball demo

Yujia-Zhang0913, 2023-10-26:
Hi, I follow the guide of tech-doc but I still cannot launch the wholebodydynamics.
The error is like the picture below. I do not know why.
[image]
Can you help me?
Moreover, sometimes the wholebodydynamics can be launched, but the gravitycompensater cannot be launched. The inertial part cannot be connect(Since now I cannot launch the dynamics, the error of gravitycompensator I cannot reproduce, I don't have the screenshot of this error)
Really thanks.

traversaro, 2023-10-26:
Hello @Yujia-Zhang0913 , this question seems to be a more general question on iCub setup, and not YARP-specific. Can you open a question in https://github.com/orgs/robotology/discussions/categories/q-a ? Thanks!

closed 2023-10-26 by Yujia-Zhang0913

Yujia-Zhang0913, 2023-10-26:
Sorry to bother you again, I want to install yarp on my own laptop. And follow the guide in this link [https://www.yarp.it/latest/install_yarp_linux.html](url), and when I check the installion, it report segment fault (core dump) like image below.(the error is reported in Chinese, in English it is segment fault (core dump) ) 
[image]
And when I run yarpserver and yarp read /portread, it is Ok. 
Thanks for helping me!

randaz81, 2023-10-26:
Dear @Yujia-Zhang0913 , I'm supposing that you installed Yarp via apt-get. 
Unfortunately, apt binary packages are currently no longer maintained (you can check the release you installed with the `yarp --version` command, the latest stable version is Yarp 3.8). So my recommendation is to build Yarp from source in order to get the latest stable version.  
However, since I imagine you are not interested in Yarp only, but in the whole robotology package, which is required to control an iCub robot, I'd advise you to follow the robotology-superbuild installation instructions: https://github.com/robotology/robotology-superbuild