robotology/yarp #3062: conda-forge based robotology-superbuild Installation facing erorr in cloning YARP

16239014, 2023-12-21:
i am doing conda-forge based robotology-superbuild Installation facing error in cloning YARP. 

**The steps i'm following here are:**
https://github.com/robotology/robotology-superbuild/blob/9761ab99b7cb1f0600d9434e032ff76da4ae7b79/doc/conda-forge.md#source-installation

**The error is:**

Cloning into 'YARP'...
error: 6829 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
Cloning into 'YARP'...
error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
error: 1628 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
Cloning into 'YARP'...
error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
error: 5357 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
-- Had to git clone more than once:
            3 times.
CMake Error at /home/sohail/robotology-superbuild/build/src/YARP/CMakeFiles/YCMTmp/YARP-gitclone.cmake:107 (message):
  Failed to clone repository: 'https://github.com/robotology/yarp.git'

gmake[2]: *** [CMakeFiles/YARP.dir/build.make:102: src/YARP/CMakeFiles/YCMStamp/YARP-download] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:947: CMakeFiles/YARP.dir/all] Error 2
gmake: *** [Makefile:101: all] Error 2

16239014, 2023-12-24:
> Hello @16239014 , can you please report:

hi @traversaro ,  sorry i forgot to mention that im using conda intead of mamba.

> * The outcome of `conda list` and `conda info` in the environment in which you are trying to install the robotology-superbuild.

**conda list**

(robsub) sohail@sohail-82TS:~$ conda list
# packages in environment at /home/sohail/Downloads/enter/envs/robsub:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
ace                       7.1.2                h59595ed_0    conda-forge
aiohttp                   3.9.1           py312h98912ed_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.10               hd590300_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
aom                       3.7.1                h59595ed_0    conda-forge
asio                      1.28.1               h59595ed_0    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
binutils                  2.40                 hdd6e379_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
binutils_linux-64         2.40                 hbdbef99_2    conda-forge
blosc                     1.21.5               h0f2a231_0    conda-forge
boost                     1.82.0               h8da182e_6    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bullet-cpp                3.25                 hfb8ada1_2    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.23.0               hd590300_0    conda-forge
c-compiler                1.7.0                hd590300_0    conda-forge
ca-certificates           2023.11.17           hbcca054_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
certifi                   2023.11.17         pyhd8ed1ab_0    conda-forge
cfitsio                   4.3.1                hbdc6101_0    conda-forge
cmake                     3.27.9               hcfe8598_0    conda-forge
compilers                 1.7.0                ha770c72_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.2.0           py312h8572e83_0    conda-forge
cppzmq                    4.10.0               h2e2a08d_1    conda-forge
cxx-compiler              1.7.0                h00ab1b0_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dartsim                   6.13.0               h7abe67e_5    conda-forge
dav1d                     1.2.1                hd590300_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
double-conversion         3.3.0                h59595ed_0    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
exceptiongroup            1.2.0              pyhd8ed1ab_0    conda-forge
executing                 2.0.1              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
fcl                       0.7.0                hadc09e8_4    conda-forge
ffmpeg                    6.1.0           gpl_ha330e6d_102    conda-forge
flann                     1.9.2                h2b5ea80_0    conda-forge
fmt                       10.1.1               h00ab1b0_1    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_1    conda-forge
fontconfig                2.14.2               h14ed4e7_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
[... 353 lines cut ...]
    conda-build version : 3.26.1
         python version : 3.11.5.final.0
       virtual packages : __archspec=1=x86_64
                          __glibc=2.38=0
                          __linux=6.5.0=0
                          __unix=0=0
       base environment : /home/sohail/Downloads/enter  (writable)
      conda av data dir : /home/sohail/Downloads/enter/etc/conda
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/sohail/Downloads/enter/pkgs
[... cut ...]
             user-agent : conda/23.7.4 requests/2.31.0 CPython/3.11.5 Linux/6.5.0-13-generic ubuntu/23.10 glibc/2.38 aau/0.4.2 c/U39swm1wQ65hrpk0lMx0Uw s/5L34VsD3QC71ib1rqduXfQ e/Lbeksv_JoxyAFnK6_8mSLQ
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False

> * The error does not seem to be actually compilation-related. If you try to run `https://github.com/robotology/yarp.git` in a newly created directory with the environment activated, does that work? Or you still get errors?
     **I** did not try this because it may effect the installation ?  I am getting error while Compiling the robotology-superbuild in cloning (downloading)

16239014, 2023-12-24:
> * The error does not seem to be actually compilation-related. If you try to run `https://github.com/robotology/yarp.git` in a newly created directory with the environment activated, does that work? Or you still get errors?
[image]

traversaro, 2023-12-24:
> > * The error does not seem to be actually compilation-related. If you try to run `https://github.com/robotology/yarp.git` in a newly created directory with the environment activated, does that work? Or you still get errors?
> >   ![Screenshot from 2023-12-24 16-32-05](https://private-user-images.githubusercontent.com/70943600/292668149-b1b0f2c8-cfe4-4b21-9a81-026dbdb13ff8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDM0MTkxNDQsIm5iZiI6MTcwMzQxODg0NCwicGF0aCI6Ii83MDk0MzYwMC8yOTI2NjgxNDktYjFiMGYyYzgtY2ZlNC00YjIxLTlhODEtMDI2ZGJkYjEzZmY4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI0VDExNTQwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ0YWU4YzEwMGExY2FjODc0MmU4NTJhMzQ1ZGU0M2Q2YTdiNDM2ZTFmMWM3OTZjYjhiNWJjMzUzZGEwNGRmNTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.095W49sEp8nNj2l3nTkEdjULdK5u4mASWSfH6N77gRo)

My bad, there was a typo in my message. What happens if you run `git clone https://github.com/robotology/yarp.git` in a clean directory?

16239014, 2023-12-24:
> My bad, there was a typo in my message. What happens if you run `git clone https://github.com/robotology/yarp.git` in a clean directory?
[image]

traversaro, 2023-12-24:
Can you try on a clean directory?

16239014, 2023-12-24:
> git clone https://github.com/robotology/yarp.git

[image]

in clean directory it cloned, but when happen when i did it earlier?

16239014, 2023-12-26:
> Can you try on a clean directory?
 I cloned YARP in clean directory then copy past in robotology package while activated environment then compile package. 
**screenshot attached**
[image]
 
**now im getiing some new type of error**
[image]

traversaro, 2023-12-27:
> **now im getiing some new type of error**

That is the error that you get if you do not install also the packages listed in https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md#source-installation, in particular the following packages:
~~~
mamba install -c conda-forge bash-completion freeglut libdc1394 libi2c expat-cos7-x86_64 libselinux-cos7-x86_64 libxau-cos7-x86_64 libxcb-cos7-x86_64 libxdamage-cos7-x86_64 libxext-cos7-x86_64 libxfixes-cos7-x86_64 libxxf86vm-cos7-x86_64 mesa-libgl-cos7-x86_64 mesa-libgl-devel-cos7-x86_64 libxshmfence-cos7-x86_64 libxshmfence-devel-cos7-x86_64 
~~~

16239014, 2024-01-04:
> > **now im getiing some new type of error**
> 
> That is the error that you get if you do not install also the packages listed in https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md#source-installation, in particular the following packages

Right now when i try to install packages im getting this type of error because installing all packages again. I change anaconda version now 3.9
[image]

traversaro, 2024-01-08:
> Right now when i try to install packages im getting this type of error because installing all packages again. I change anaconda version now 3.9

Which type of error? I do not see any error in your issue. In general, please try to report error as text and not as screenshot, thanks!

16239014, 2024-03-04:
cn

closed 2024-03-04 by 16239014