robotology/yarp #3063: how can I resolve the "Reply from iCubSim: [fail] "could not set index" error"?

bionicsgirl, 2023-12-21:
Hi there!
I have to create a thread of a sphere that moves first on the right to reach the end of a radius in 2 seconds of time and then move in circle with a frequency of 0.2 Hz in Yarp and iCub simulator programming in C++.
This is my code and, after initializing the sphere correctly i get this error: Reply from iCubSim: [fail] "could not set index"
How can i resolve this?
Thank you in advance!
```
#include <iostream>
#include <yarp/os/all.h>
#include <yarp/sig/all.h>
#include <yarp/dev/all.h>

using namespace yarp::os;
using namespace yarp::dev;
using namespace yarp::sig;
using namespace std;

class SphereThread: public PeriodicThread {

	public: 

             SphereThread(double p):PeriodicThread(p){
			
			bool ok  = sphere_port.open("/sphere-client");
				if (!ok) {
					std::cerr << "Failed to open port" << std::endl;
					return;
				}
			// Connect the client to the RPC Server of iCub_SIM
			yarp.connect("/sphere-client", "/icubSim/world");
			}
		
		

	protected:
    
    	    bool threadInit(){
			
			
			req.addString("world");
			req.addString("mk");
			req.addString("ssph");
			req.addFloat64(0.04);
			req.addFloat64(0.02);
			req.addFloat64(0.9);
			req.addFloat64(0.8);
			req.addFloat64(1);
			req.addFloat64(0);
			req.addFloat64(0);  
			sphere_port.write(req,reply);
			cout << "Reply from iCubSim: " << reply.toString() << endl;
   	    	        return true;
			
			
   	    }

		void run() {
			double initial_time = Time::now();
				while (Time::now() - initial_time < 2.0){
					double t = Time::now() - initial_time;
					double xPos = 0.02 + 0.4*(t/2.0);
					double yPos = 0.9;
					setPosition(xPos, yPos);
				}
				while (Time::now() < initial_time+2) {

					double t = Time::now() - initial_time;
					double theta = 2.0*M_PI*0.2*t;
					double xPos = 0.02 + 0.4*cos(theta);
					double yPos = 0.9  + 0.4*sin(theta);
					setPosition(xPos, yPos);
				
				}

			}
			
		
			
		
		void threadRelease() {
			yarp.disconnect("/sphere-client", "/icubSim/world");
    		        sphere_port.close();
			cout << "Done, goodbye from myThread." << endl;
		}
	
    private:

		Network yarp;  //initialize the network
    	        RpcClient sphere_port;  // create rpc client for the sphere
		Bottle req, reply;
		
		
		void setPosition(double x, double y){
			req.clear();
			req.addString("world");
			req.addString("set");
			req.addString("ssph");
			req.addFloat64(1);
			req.addFloat64(x);
			req.addFloat64(y);
			req.addFloat64(0.8);
			sphere_port.write(req, reply);
			cout << "Reply from iCubSim: " << reply.toString() << endl;
		}
        

};

int main(int argc, char *argv[]) {

    Network yarp;
    srand(time(NULL));
    SphereThread mt(0.2/0.1);
    mt.start();
    
    Time::delay(20);
    mt.stop();

    return 0;

};

```

pattacini, 2023-12-21:
Hi @bionicsgirl 

You already filed a discussion on the same topic.
See https://github.com/orgs/robotology/discussions/677.

Please, avoid double posting.

cc @randaz81

bionicsgirl, 2023-12-21:
okay, i'm sorry, but thank you!

closed 2023-12-21 by bionicsgirl