robotology/yarp #3067: Failure to compile YARP with the option `ENABLE_yarpcar_portmonitor`, ON

EhsanRanjbari, 2024-01-08:
I am having a compilation error as below when the option `ENABLE_yarpcar_portmonitor` is `ON`: (https://github.com/ami-iit/element_haptic-gloves/issues/159#issuecomment-1805295330)

```bash
[ 50%] Building CXX object src/portmonitors/image_compression_ffmpeg/CMakeFiles/yarp_pm_image_compression_ffmpeg.dir/ffmpegPortmonitor.cpp.o
[ 50%] Linking CXX shared module ../../../lib/yarp/yarp_pm_image_compression_ffmpeg.so
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libswscale.a(hscale_fast_bilinear_simd.o): relocation R_X86_64_32S against `.text.unlikely' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: failed to set dynamic section sizes: bad value
collect2: error: ld returned 1 exit status
make[5]: *** [src/portmonitors/image_compression_ffmpeg/CMakeFiles/yarp_pm_image_compression_ffmpeg.dir/build.make:118: lib/yarp/yarp_pm_image_compression_ffmpeg.so] Error 1
make[4]: *** [CMakeFiles/Makefile2:5623: src/portmonitors/image_compression_ffmpeg/CMakeFiles/yarp_pm_image_compression_ffmpeg.dir/all] Error 2
make[3]: *** [Makefile:156: all] Error 2
make[2]: *** [CMakeFiles/YARP.dir/build.make:85: src/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:1083: CMakeFiles/YARP.dir/all] Error 2
make: *** [Makefile:101: all] Error 2
```

**Configuration (please complete the following information):**
 - OS: Ubuntu 22.04.3 LTS.
 - yarp version: 3.9

see: https://github.com/robotology/yarp/issues/2885#issuecomment-1876782445

traversaro, 2024-01-08:
Can you report the output of `apt list --installed` and the content of YARP's CMakeCache.txt ?

EhsanRanjbari, 2024-01-09:
> Can you report the output of `apt list --installed` and the content of YARP's CMakeCache.txt ?

`apt list --installed`: 
[apt_list_installed.txt](https://github.com/robotology/yarp/files/13871334/apt_list_installed.txt)

YARP's CMakeCache.txt:
[CMakeCache.txt](https://github.com/robotology/yarp/files/13871335/CMakeCache.txt)

traversaro, 2024-01-09:
Ok, the strange part of the CMakeCache is this one:
~~~
//FFMPEG's avcodec include directory
FFMPEG_avcodec_INCLUDE_DIR:PATH=/usr/include/x86_64-linux-gnu

//FFMPEG's avcodec library
FFMPEG_avcodec_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavcodec.so

//FFMPEG's avdevice include directory
FFMPEG_avdevice_INCLUDE_DIR:PATH=/usr/include/x86_64-linux-gnu

//FFMPEG's avdevice library
FFMPEG_avdevice_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavdevice.so

//FFMPEG's avfilter include directory
FFMPEG_avfilter_INCLUDE_DIR:PATH=/usr/include/x86_64-linux-gnu

//FFMPEG's avfilter library
FFMPEG_avfilter_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavfilter.so

//FFMPEG's avformat include directory
FFMPEG_avformat_INCLUDE_DIR:PATH=/usr/include/x86_64-linux-gnu

//FFMPEG's avformat library
FFMPEG_avformat_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavformat.so

//FFMPEG's avresample include directory
FFMPEG_avresample_INCLUDE_DIR:PATH=FFMPEG_avresample_INCLUDE_DIR-NOTFOUND

//FFMPEG's avresample library
FFMPEG_avresample_LIBRARY:FILEPATH=FFMPEG_avresample_LIBRARY-NOTFOUND

//FFMPEG's avutil include directory
FFMPEG_avutil_INCLUDE_DIR:PATH=/usr/include/x86_64-linux-gnu

//FFMPEG's avutil library
FFMPEG_avutil_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libavutil.so

//FFMPEG's swresample include directory
FFMPEG_swresample_INCLUDE_DIR:PATH=/usr/include/x86_64-linux-gnu

//FFMPEG's swresample library
FFMPEG_swresample_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libswresample.so

//FFMPEG's swscale include directory
FFMPEG_swscale_INCLUDE_DIR:PATH=/usr/include/x86_64-linux-gnu

//FFMPEG's swscale library
FFMPEG_swscale_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libswscale.a
~~~

The `FFMPEG_swscale_LIBRARY` variable should actually be `/usr/lib/x86_64-linux-gnu/libswscale.so`, and I have no idea why instead it is set to `/usr/lib/x86_64-linux-gnu/libswscale.a`. Both files are part of the same Debian package `libswscale-dev`, see:
~~~
traversaro@IITICUBLAP257:~$ apt-file search /usr/lib/x86_64-linux-gnu/libswscale.a
libswscale-dev: /usr/lib/x86_64-linux-gnu/libswscale.a
traversaro@IITICUBLAP257:~$ apt-file search /usr/lib/x86_64-linux-gnu/libswscale.so
libswscale-dev: /usr/lib/x86_64-linux-gnu/libswscale.so
libswscale5: /usr/lib/x86_64-linux-gnu/libswscale.so.5
libswscale5: /usr/lib/x86_64-linux-gnu/libswscale.so.5.9.100
~~~

I wonder if somehow `/usr/lib/x86_64-linux-gnu/libswscale.so` is corrupted for some reason on your machine? What is the output of `file /usr/lib/x86_64-linux-gnu/libswscale.so` and of `file /usr/lib/x86_64-linux-gnu/libswscale.so.
5.9.100`?

If the `/usr/lib/x86_64-linux-gnu/libswscale.so` file exists, can you try to delete the YARP build cache and re-configure and re-compile YARP, i.e. : 

~~~
cd /home/Ehsan/robotology-superbuild/
rm -rf ./build/src/YARP
make YARP
~~~
?

If the problem persists, I would be curious to debug the issue.

EhsanRanjbari, 2024-01-09:
> What is the output of `file /usr/lib/x86_64-linux-gnu/libswscale.so`

```bash
Ehsan@iiticublap281lu:~$ file /usr/lib/x86_64-linux-gnu/libswscale.so
/usr/lib/x86_64-linux-gnu/libswscale.so: cannot open `/usr/lib/x86_64-linux-gnu/libswscale.so' (No such file or directory)
```

> file /usr/lib/x86_64-linux-gnu/libswscale.so. 5.9.100
```bash
Ehsan@iiticublap281lu:~$ file /usr/lib/x86_64-linux-gnu/libswscale.so. 5.9.100
/usr/lib/x86_64-linux-gnu/libswscale.so.: cannot open `/usr/lib/x86_64-linux-gnu/libswscale.so.' (No such file or directory)
5.9.100:                                  cannot open `5.9.100' (No such file or directory)
```

traversaro, 2024-01-09:
Bingo! For some reason `/usr/lib/x86_64-linux-gnu/libswscale.so` is not there, I have no idea why. Can you try to re-install `libswscale-dev`?

traversaro, 2024-01-09:
> > What is the output of `file /usr/lib/x86_64-linux-gnu/libswscale.so`
> 
> ```shell
> Ehsan@iiticublap281lu:~$ file /usr/lib/x86_64-linux-gnu/libswscale.so
> /usr/lib/x86_64-linux-gnu/libswscale.so: cannot open `/usr/lib/x86_64-linux-gnu/libswscale.so' (No such file or directory)
> ```
> 
> > file /usr/lib/x86_64-linux-gnu/libswscale.so. 5.9.100
> 
> ```shell
> Ehsan@iiticublap281lu:~$ file /usr/lib/x86_64-linux-gnu/libswscale.so. 5.9.100
> /usr/lib/x86_64-linux-gnu/libswscale.so.: cannot open `/usr/lib/x86_64-linux-gnu/libswscale.so.' (No such file or directory)
> 5.9.100:                                  cannot open `5.9.100' (No such file or directory)
> ```

Sorry, I did a typo, the correct command was `file /usr/lib/x86_64-linux-gnu/libswscale.so.5.9.100`.

EhsanRanjbari, 2024-01-09:
> Sorry, I did a typo, the correct command was `file /usr/lib/x86_64-linux-gnu/libswscale.so.5.9.100`.
```bash
Ehsan@iiticublap281lu:~$ file /usr/lib/x86_64-linux-gnu/libswscale.so.5.9.100
/usr/lib/x86_64-linux-gnu/libswscale.so.5.9.100: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=a9a5e85a05918ececa64eb96fc88599b127990c3, stripped
```

traversaro, 2024-01-09:
Interesting, so only `/usr/lib/x86_64-linux-gnu/libswscale.so` . At this point I would try to reinstall `libswscale-dev` to see if it ends being installed.

EhsanRanjbari, 2024-01-09:
> I would try to reinstall `libswscale-dev` to see if it ends being installed.

I did try to install but nothing installed.

traversaro, 2024-01-09:
> > I would try to reinstall `libswscale-dev` to see if it ends being installed.
> 
> I did try to install but nothing was installed.

Which command did you run? To re-install the package, you need to run `sudo apt install --reinstall libswscale-dev`.

EhsanRanjbari, 2024-01-09:
> > > I would try to reinstall `libswscale-dev` to see if it ends being installed.
> > 
> > 
> > I did try to install but nothing was installed.
> 
> Which command did you run? To re-install the package, you need to run `sudo apt install --reinstall libswscale-dev`.

ah, ok I did `sudo apt-get install -y libswscale-dev`, but let me use your command.

EhsanRanjbari, 2024-01-09:
After running the `sudo apt install --reinstall libswscale-dev`, it installed it. and here is the output for the commands above:
`file /usr/lib/x86_64-linux-gnu/libswscale.so`:
```bash
/usr/lib/x86_64-linux-gnu/libswscale.so: symbolic link to libswscale.so.5.9.100
```

`file /usr/lib/x86_64-linux-gnu/libswscale.so.5.9.100`:
```bash
/usr/lib/x86_64-linux-gnu/libswscale.so.5.9.100: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=a9a5e85a05918ececa64eb96fc88599b127990c3, stripped
```

Now, I am gonna recompile to see if the issue got fixed.

traversaro, 2024-01-09:
> Now, I am gonna recompile to see if the issue got fixed.

You probably need to start from a clean build, or at least remove the `FFMPEG_swscale_LIBRARY` variable from YARP's CMake cache (or if you want to do quick, just change it from .a to .so. This because the cache is created only once, and if you just re-run CMake the cached values for the variables are used.

traversaro, 2024-01-09:
By the way, do you have any idea of any operation that you did on your computer (with `sudo` permission) that could have removed `/usr/lib/x86_64-linux-gnu/libswscale.so`?

EhsanRanjbari, 2024-01-09:
> You probably need to start from a clean build, or at least remove the `FFMPEG_swscale_LIBRARY` variable from YARP's CMake cache (or if you want to do quick, just change it from .a to .so. This [...]

Thanks. Before seeing this command I did reinstall it successfully without having the mentioned issue. Thanks! I will also do the clean install.

traversaro, 2024-01-09:
Ok, feel free to close the issue then!

EhsanRanjbari, 2024-01-09:
> By the way, do you have any idea of any operation that you did on your computer (with `sudo` permission) that could have removed `/usr/lib/x86_64-linux-gnu/libswscale.so`?

I may recall something related to the scaling. I had an issue with the latest Ubuntu versions which when connecting to a second display, the resolution was high but the icons and tabs were tiny. To solve it I installed https://gitlab.gnome.org/GNOME/gnome-tweaks. It was a long time ago and I may have done some other stuff as well which I do not recall. I do not know if that could be the reason.

traversaro, 2024-01-09:
It is difficult to tell, but thanks for reporting this!

traversaro, 2024-01-09:
@EhsanRanjbari if you think the problem is solved, feel free to close the issue.

EhsanRanjbari, 2024-01-10:
Closing the issue as completed. Thanks @traversaro

closed 2024-01-10 by EhsanRanjbari