robotology/yarp #3083: issue with swig 4.2.0

randaz81, 2024-02-09:
Swig 4.2.0 seems not to work properly.
The recommended version to use is thus 4.0.2

traversaro, 2024-02-09:
Did you tried with swig master? I think the fix for https://github.com/swig/swig/issues/2783 should have fixed yarp generation, but perhaps I was testing an earlier version of YARP.

randaz81, 2024-09-16:
It looks like that Ubuntu 24.04 is currently including by default swig 4.2.0

randaz81, 2024-09-17:
Thank you @traversaro , I'll follow your suggestion for Yarp CI with Ubuntu 24.04

PeterBowman, 2024-10-01:
On a side note, it seems that SWIG 4.2.0 is preinstalled on Ubuntu 24.04 runners for GitHub Actions. I had to remove the "swig" package via apt before building it from sources.

randaz81, 2025-01-21:
in CI this issue was fixed by using SWIG version 4.2.1, see https://github.com/robotology/yarp/pull/3138