robotology/yarp #3087: conda-forge based robotology-superbuild  is not running icub simulator

16239014, 2024-03-04:
**I am facing error while running icub simulator:**

(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~/tutorial_joint-interface$ cmake -S . -B build
CMake Deprecation Warning at CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Error at CMakeLists.txt:8 (find_package):
  By not providing "FindYARP.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "YARP", but
  CMake did not find one.

  Could not find a package configuration file provided by "YARP" (requested
  version 3.5.1) with any of the following names:

    YARPConfig.cmake
    yarp-config.cmake

  Add the installation prefix of "YARP" to CMAKE_PREFIX_PATH or set
  "YARP_DIR" to a directory containing one of the above files.  If "YARP"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!

traversaro, 2024-03-04:
Hello @16239014 , can you please describe how you installed the robotology-superbuild? Did you remember to source the setup.sh of the superbuild before running the tutorial?

16239014, 2024-03-05:
> Hello @16239014 , can you please describe how you installed the robotology-superbuild? Did you remember to source the setup.sh of the superbuild before running the tutorial?

yes i did source the setup.sh then i run yarpserver after that i run a project 
[https://icub-tech-iit.github.io/documentation/sw_installation/check_your_installation/](url)

16239014, 2024-03-05:
> Can you report the output of the env command in the terminal where you are running cmake?

(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~/tutorial_joint-interface$ yarpserver --write

     ==========================
    | __  __ ___  ____   ____  |
    | \ \/ //   ||  _ \ |  _ \ |
    |  \  // /| || |/ / | |/ / |
    |  / // ___ ||  _ \ |  _/  |
    | /_//_/  |_||_| \_\|_|    |
    |                          |
     ==========================

Call with --help for information on available options
Using port database: :memory:
Using subscription database: :memory:
IP address: default
Port number: 10000
Overriding non-local address for name server
[INFO] |yarp.os.Port|/root| Port /root active at tcp://192.168.100.26:10000/
Registering name server with itself
 * register "/root" tcp "192.168.100.26" 10000
   + set "/root" offers http name_ser local tcp fast_tcp mcast udp text text_ack bayer human mjpeg portmonitor priority rossrv shmem tcpros unix_stream xmlrpc
   + set "/root" accepts http name_ser local tcp fast_tcp mcast udp text text_ack bayer human mjpeg portmonitor priority rossrv shmem tcpros unix_stream xmlrpc
   + set "/root" ips "127.0.0.1" "192.168.100.26"
   + set "/root" process 37364
 * register fallback mcast "224.2.1.1" 10000
   + set fallback offers http name_ser local tcp fast_tcp mcast udp text text_ack bayer human mjpeg portmonitor priority rossrv shmem tcpros unix_stream xmlrpc
   + set fallback accepts http name_ser local tcp fast_tcp mcast udp text text_ack bayer human mjpeg portmonitor priority rossrv shmem tcpros unix_stream xmlrpc
   + set fallback ips "127.0.0.1" "192.168.100.26"
   + set fallback process 37364
 * set "/root" nameserver "true"
Name server can be browsed at http://192.168.100.26:10000/
Ok. Ready!

16239014, 2024-03-05:
> Can you report the output of the env command in the terminal where you are running cmake?

(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~/tutorial_joint-interface$ env
SHELL=/bin/bash
DEBUG_FFLAGS=-march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/sohail/anaconda3/envs/robsub/include -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-all -fno-plt -Og -g -Wall -Wextra -fcheck=all -fbacktrace -fimplicit-none -fvar-tracking-assignments -ffunction-sections -pipe
SESSION_MANAGER=local/sohail-Lenovo-V14-G3-IAP:@/tmp/.ICE-unix/2135,unix/sohail-Lenovo-V14-G3-IAP:/tmp/.ICE-unix/2135
GAZEBO_MASTER_URI=http://localhost:11345
QT_ACCESSIBILITY=1
COLORTERM=truecolor
GCC_RANLIB=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-gcc-ranlib
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg
XDG_MENU_PREFIX=gnome-
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
CONDA_EXE=/home/sohail/anaconda3/bin/conda
_CE_M=
ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX=/home/sohail/robotology-superbuild/build/install
build_alias=x86_64-conda-linux-gnu
CMAKE_ARGS=-DCMAKE_AR=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-ar -DCMAKE_CXX_COMPILER_AR=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-gcc-ar -DCMAKE_C_COMPILER_AR=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-gcc-ar -DCMAKE_RANLIB=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-ranlib -DCMAKE_CXX_COMPILER_RANLIB=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-gcc-ranlib -DCMAKE_C_COMPILER_RANLIB=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-gcc-ranlib -DCMAKE_LINKER=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-ld -DCMAKE_STRIP=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-strip -DCMAKE_BUILD_TYPE=Release
GPROF=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-gprof
CONDA_TOOLCHAIN_BUILD=x86_64-conda-linux-gnu
GNOME_SHELL_SESSION_MODE=ubuntu
_CONDA_PYTHON_SYSCONFIGDATA_NAME=_sysconfigdata_x86_64_conda_cos6_linux_gnu
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
STRINGS=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-strings
CPP=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-cpp
XMODIFIERS=@im=ibus
DESKTOP_SESSION=ubuntu
SSH_AGENT_PID=2087
ROS_PACKAGE_PATH=/home/sohail/robotology-superbuild/build/install/share
XML_CATALOG_FILES=file:///home/sohail/anaconda3/envs/robsub/etc/xml/catalog file:///etc/xml/catalog
FFLAGS=-march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/sohail/anaconda3/envs/robsub/include
GTK_MODULES=gail:atk-bridge
PWD=/home/sohail/tutorial_joint-interface
GSETTINGS_SCHEMA_DIR=/home/sohail/anaconda3/envs/robsub/share/glib-2.0/schemas
LOGNAME=sohail
XDG_SESSION_DESKTOP=ubuntu
XDG_SESSION_TYPE=x11
CONDA_PREFIX=/home/sohail/anaconda3/envs/robsub
GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1
GSETTINGS_SCHEMA_DIR_CONDA_BACKUP=
CXX=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-c++
CXXFLAGS=-fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/sohail/anaconda3/envs/robsub/include
XAUTHORITY=/run/user/1000/gdm/Xauthority
CONDA_TOOLCHAIN_HOST=x86_64-conda-linux-gnu
DEBUG_CXXFLAGS=-fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-all -fno-plt -Og -g -Wall -Wextra -fvar-tracking-assignments -ffunction-sections -pipe -isystem /home/sohail/anaconda3/envs/robsub/include
GJS_DEBUG_TOPICS=JS ERROR;JS LOG
WINDOWPATH=2
LDFLAGS=-Wl,-O2 -Wl,--sort-common -Wl,--as-needed -Wl,-z,relro -Wl,-z,now -Wl,--disable-new-dtags -Wl,--gc-sections -Wl,--allow-shlib-undefined -Wl,-rpath,/home/sohail/anaconda3/envs/robsub/lib -Wl,-rpath-link,/home/sohail/anaconda3/envs/robsub/lib -L/home/sohail/anaconda3/envs/robsub/lib
HOME=/home/sohail
USERNAME=sohail
IM_CONFIG_PHASE=1
GAZEBO_PLUGIN_PATH=/home/sohail/anaconda3/envs/robsub/lib/gazebo-11/plugins::/home/sohail/robotology-superbuild/build/install/lib
LANG=en_US.UTF-8
MESON_ARGS=--buildtype release
GAZEBO_MODEL_DATABASE_URI=http://models.gazebosim.org
DEBUG_CFLAGS=-march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-all -fno-plt -Og -g -Wall -Wextra -fvar-tracking-assignments -ffunction-sections -pipe -isystem /home/sohail/anaconda3/envs/robsub/include
XDG_CURRENT_DESKTOP=ubuntu:GNOME
VTE_VERSION=6003
CXX_FOR_BUILD=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-c++
ELFEDIT=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-elfedit
CONDA_PROMPT_MODIFIER=(robsub) 
AMENT_PREFIX_PATH=/home/sohail/robotology-superbuild/build/install
CMAKE_PREFIX_PATH=/home/sohail/anaconda3/envs/robsub:/home/sohail/anaconda3/envs/robsub/x86_64-conda-linux-gnu/sysroot/usr:/home/sohail/robotology-superbuild/build/install
GNOME_TERMINAL_SCREEN=/org/gnome/Terminal/screen/22794167_3301_4258_bef2_3424ba248d90
CPPFLAGS=-DNDEBUG -D_FORTIFY_SOURCE=2 -O2 -isystem /home/sohail/anaconda3/envs/robsub/include
CPL_ZIP_ENCODING=UTF-8
GAZEBO_RESOURCE_PATH=/home/sohail/anaconda3/envs/robsub/share/gazebo-11::/home/sohail/robotology-superbuild/build/install/share/gazebo/worlds
LD=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-ld
INVOCATION_ID=40615cb8cd02422eac67e2561bfefe19
MANAGERPID=1900
[... cut ...]
CFLAGS=-march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/sohail/anaconda3/envs/robsub/include
CXXFILT=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-c++filt
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
BUILD=x86_64-conda-linux-gnu
CONDA_PREFIX_1=/home/sohail/anaconda3
RANLIB=/home/sohail/anaconda3/envs/robsub/bin/x86_64-conda-linux-gnu-ranlib
CONDA_BUILD_SYSROOT=/home/sohail/anaconda3/envs/robsub/x86_64-conda-linux-gnu/sysroot
PROJ_NETWORK=ON
OGRE_RESOURCE_PATH=/home/sohail/anaconda3/envs/robsub/lib/OGRE
OLDPWD=/home/sohail
_=/usr/bin/env

traversaro, 2024-03-05:
Can you try on a clean build? I guess something is going wrong in the install prefix set by the icubcontrib package. Alternatively, you can just set yourself a CMAKE_INSTALL_PREFIX different from `/usr/local` .

16239014, 2024-03-05:
> Can you try on a clean build? I guess something is going wrong in the install prefix set by the icubcontrib package. Alternatively, you can just set yourself a CMAKE_INSTALL_PREFIX different from `/usr/local` .

yes i have tried clean build. 
how to do this ???(you can just set yourself a CMAKE_INSTALL_PREFIX different from `/usr/local`)

traversaro, 2024-03-05:
> how to do this ???(you can just set yourself a CMAKE_INSTALL_PREFIX different from `/usr/local`)

Just pass `-DCMAKE_INSTALL_PREFIX=/home/sohail/robotology-superbuild/build/install` to the `cmake -S . -B build` command.

16239014, 2024-03-05:
> -DCMAKE_INSTALL_PREFIX=/home/sohail/robotology-superbuild/build/install

(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~/tutorial_joint-interface$ -DCMAKE_INSTALL_PREFIX=/home/sohail/robotology-superbuild/build/install
bash: -DCMAKE_INSTALL_PREFIX=/home/sohail/robotology-superbuild/build/install: No such file or directory

traversaro, 2024-03-05:
`-DCMAKE_INSTALL_PREFIX=/home/sohail/robotology-superbuild/build/install` is an option, not a command on its own. You need to call:
~~~
cmake -S . -B build -DCMAKE_INSTALL_PREFIX=/home/sohail/robotology-superbuild/build/install
~~~

16239014, 2024-03-05:
yes it worked. Thanks!

16239014, 2024-03-05:
> `-DCMAKE_INSTALL_PREFIX=/home/sohail/robotology-superbuild/build/install` is an option, not a command on its own. You need to call:
> 
> ```
> cmake -S . -B build -DCMAKE_INSTALL_PREFIX=/home/sohail/robotology-superbuild/build/install
> ```

so whenever i want to run any project i must have run this command first??

16239014, 2024-03-11:
**i am installing again that example and error occur at installation**

(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~/tutorial_joint-interface$ cmake -S . -B build -DCMAKE_INSTALL_PREFIX=/home/sohail/robotology-superbuild/build/install
CMake Deprecation Warning at CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found YCM: /home/sohail/robotology-superbuild/build/install/share/cmake/YCM (found version "0.16.2.3-20240214.3+gitadccba6")
-- Found YARP: /home/sohail/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.9.0+7-20240123.1+gitee0524166")
-- Found YARP: /home/sohail/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.9.0+7-20240123.1+gitee0524166")
-- Libraries are placed in /home/sohail/tutorial_joint-interface/build/lib
-- Executables are placed in /home/sohail/tutorial_joint-interface/build/bin
-- Configuring done (0.7s)
-- Generating done (0.0s)
-- Build files have been written to: /home/sohail/tutorial_joint-interface/build

**after that when i enter:**cmake --build build --target install

collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/tutorial_joint-interface-position.dir/build.make:102: bin/tutorial_joint-interface-position] Error 1
make[1]: *** [CMakeFiles/Makefile2:105: CMakeFiles/tutorial_joint-interface-position.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~/tutorial_joint-interface$ 

**when i reinstall the [**](https://icub-tech-iit.github.io/documentation/sw_installation/check_your_installation/#check-yarp) **again there are errors **

traversaro, 2024-03-11:
Please provide the complete error log, not just the last lines.

16239014, 2024-03-18:
> Please provide the complete error log, not just the last lines.
 **full error log is** 
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~/tutorial_joint-interface$ cmake --build build --target install
[ 16%] Linking CXX executable bin/tutorial_joint-interface-position
/home/sohail/anaconda3/envs/robsub/bin/ld: warning: libz.so.1, needed by /home/sohail/robotology-superbuild/build/install/lib/libYARP_dev.so.3.9.0, not found (try using -rpath or -rpath-link)
/home/sohail/anaconda3/envs/robsub/bin/ld: warning: libACE.so.7.1.3, needed by /home/sohail/robotology-superbuild/build/install/lib/libYARP_os.so.3.9.0, not found (try using -rpath or -rpath-link)
/home/sohail/anaconda3/envs/robsub/bin/ld: warning: libedit.so.0, needed by /home/sohail/robotology-superbuild/build/install/lib/libYARP_os.so.3.9.0, not found (try using -rpath or -rpath-link)
/home/sohail/anaconda3/envs/robsub/bin/ld: warning: libjpeg.so.8, needed by /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0, not found (try using -rpath or -rpath-link)
/home/sohail/anaconda3/envs/robsub/bin/ld: warning: libsoxr.so.0, needed by /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0, not found (try using -rpath or -rpath-link)
/home/sohail/anaconda3/envs/robsub/bin/ld: warning: libpng16.so.16, needed by /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0, not found (try using -rpath or -rpath-link)
/home/sohail/anaconda3/envs/robsub/bin/ld: warning: libavcodec.so.60, needed by /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0, not found (try using -rpath or -rpath-link)
/home/sohail/anaconda3/envs/robsub/bin/ld: warning: libavutil.so.58, needed by /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0, not found (try using -rpath or -rpath-link)
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0: undefined reference to `av_frame_make_writable@LIBAVUTIL_58'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0: undefined reference to `avcodec_send_frame@LIBAVCODEC_60'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0: undefined reference to `png_create_read_struct@PNG16_0'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_os.so.3.9.0: undefined reference to `ACE_DLL::close()'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_os.so.3.9.0: undefined reference to `ACE_INET_Addr::get_ip_address() const'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_os.so.3.9.0: undefined reference to `vtable for ACE_Time_Value'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0: undefined reference to `av_frame_get_buffer@LIBAVUTIL_58'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0: undefined reference to `jpeg_std_error@LIBJPEG_8.0'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_os.so.3.9.0: undefined reference to `ACE_DLL::~ACE_DLL()'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_dev.so.3.9.0: undefined reference to `compress'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0: undefined reference to `avcodec_free_context@LIBAVCODEC_60'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_os.so.3.9.0: undefined reference to `ACE_Stack_Trace::c_str() const'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0: undefined reference to `av_get_sample_fmt_name@LIBAVUTIL_58'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0: undefined reference to `png_write_image@PNG16_0'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_os.so.3.9.0: undefined reference to `ACE_SOCK_Acceptor::ACE_SOCK_Acceptor()'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0: undefined reference to `avcodec_alloc_context3@LIBAVCODEC_60'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_os.so.3.9.0: undefined reference to `ACE::send_n_i(int, void const*, unsigned long, ACE_Time_Value const*, unsigned long*)'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_os.so.3.9.0: undefined reference to `ACE_SOCK::close()'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0: undefined reference to `jpeg_read_header@LIBJPEG_8.0'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_dev.so.3.9.0: undefined reference to `uncompress'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0: undefined reference to `jpeg_set_defaults@LIBJPEG_8.0'
[... cut ...]
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_os.so.3.9.0: undefined reference to `ACE_DLL::error() const'
/home/sohail/anaconda3/envs/robsub/bin/ld: /home/sohail/robotology-superbuild/build/install/lib/libYARP_sig.so.3.9.0: undefined reference to `av_parser_parse2@LIBAVCODEC_60'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/tutorial_joint-interface-position.dir/build.make:102: bin/tutorial_joint-interface-position] Error 1
make[1]: *** [CMakeFiles/Makefile2:105: CMakeFiles/tutorial_joint-interface-position.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

traversaro, 2024-03-18:
~~~
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~/tutorial_joint-interface$ cmake --build build --target install
[ 16%] Linking CXX executable bin/tutorial_joint-interface-position
/home/sohail/anaconda3/envs/robsub/bin/ld: warning: libz.so.1, needed by /home/sohail/robotology-superbuild/build/install/lib/libYARP_dev.so.3.9.0, not found (try using -rpath or -rpath-link)
~~~

This error is quite strange, it is like the libraries required by YARP are not anymore there. Can you report the output of `mamba list`?

16239014, 2024-03-18:
> ```
> (robsub) sohail@sohail-Lenovo-V14-G3-IAP:~/tutorial_joint-interface$ cmake --build build --target install
> [ 16%] Linking CXX executable bin/tutorial_joint-interface-position [...]

(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~/tutorial_joint-interface$ conda list
# packages in environment at /home/sohail/anaconda3/envs/robsub:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
_sysroot_linux-64_curr_repodata_hack 3                   h69a702a_14    conda-forge
ace                       7.1.3                h59595ed_2    conda-forge
aiohttp                   3.9.3           py312h98912ed_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.10               hd590300_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
aom                       3.7.1                h59595ed_0    conda-forge
asio                      1.29.0               h59595ed_0    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
azure-core-cpp            1.10.3               h91d86a7_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h00ab1b0_0    conda-forge
azure-storage-common-cpp  12.5.0               hb858b4b_2    conda-forge
bash-completion           2.11                 ha770c72_1    conda-forge
binutils                  2.40                 hdd6e379_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
binutils_linux-64         2.40                 hbdbef99_2    conda-forge
blosc                     1.21.5               h0f2a231_0    conda-forge
boost                     1.82.0               h8da182e_6    conda-forge
boost-cpp                 1.82.0               h44aadfe_6    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bullet-cpp                3.25                 hfb8ada1_2    conda-forge
bzip2                     1.0.8                hd590300_5    conda-forge
c-ares                    1.26.0               hd590300_0    conda-forge
c-compiler                1.7.0                hd590300_0    conda-forge
ca-certificates           2024.2.2             hbcca054_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
cfitsio                   4.3.0                hbdc6101_0    conda-forge
cmake                     3.28.3               hcfe8598_0    conda-forge
compilers                 1.7.0                ha770c72_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.2.0           py312h8572e83_0    conda-forge
cppzmq                    4.10.0               h2e2a08d_1    conda-forge
cxx-compiler              1.7.0                h00ab1b0_0    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
dartsim                   6.13.1               h7abe67e_0    conda-forge
dav1d                     1.2.1                hd590300_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
double-conversion         3.3.0                h59595ed_0    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
exceptiongroup            1.2.0              pyhd8ed1ab_2    conda-forge
executing                 2.0.1              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
expat-cos7-x86_64         2.1.0             h9b0a68f_1105    conda-forge
fcl                       0.7.0                hadc09e8_4    conda-forge
ffmpeg                    6.1.1           gpl_h186bccc_100    conda-forge
flann                     1.9.2                h2b5ea80_0    conda-forge
fmt                       10.2.1               h00ab1b0_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 h77eed37_1    conda-forge
[... 350 lines cut ...]
xorg-libxcursor           1.2.0                h0b41bf4_1    conda-forge
xorg-libxdamage           1.1.5                h7f98852_1    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h0b41bf4_2    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxinerama          1.1.5                h27087fc_0    conda-forge
[... cut ...]
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yarl                      1.9.4           py312h98912ed_0    conda-forge
zeromq                    4.3.5                h59595ed_0    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge

traversaro, 2024-03-18:
This is quite strange, that file should be provided by the `libzlib` package that is there. Can you check that the `libz.so.1` is indeed contained in `$CONDA_PREFIX\lib` ?

Other useful information may be:
* Output of `env` 
* Verbose output of the compilation

In general, please use `~~~` and `~~~` code blocks to include logs in comments.

traversaro, 2024-03-22:
@16239014 given the lack of feedback, I would close the issue here. As the problem is more related to robotology-superbuild, feel free to open an issue in https://github.com/robotology/robotology-superbuild/ if you still have problems.

closed 2024-03-22 by traversaro