robotology/yarp #3096: NetworkClock not working in with LocalMode enabled

xela-95, 2024-03-20:
**Describe the bug**
Hi, I've done a unit test, part of [gz-sim-yarp-plugins](https://github.com/robotology/gz-sim-yarp-plugins/tree/main) in which I want to test the functionality of the clock plugin, by instantiating a `yarp::os::NetworkClock`, opening it on the `/clock` port and then calling the `now()` method to get the current time published on the port. 

Since this is a unit test, I need to do this locally, hence having done `yarp::os::NetworkBase::setLocalMode(true);`.

When I launch the test I get the following errors:

```
[ERROR] |yarp.os.Network| cannot find port /root
Failure: name server did not accept connection to topic.
```

**To Reproduce**
Here's the code of my unit test:

```cpp
#include <gtest/gtest.h>
#include <gz/sim/TestFixture.hh>

#include <yarp/os/Bottle.h>
#include <yarp/os/BufferedPort.h>
#include <yarp/os/Network.h>

TEST(ClockTest, ClockPluginPublishTimeCoherentWithSimTime)
{
    // ARRANGE
    yarp::os::NetworkBase::setLocalMode(true);
    // Maximum verbosity helps with debugging
    gz::common::Console::SetVerbosity(4);
    // Instantiate test fixture
    gz::sim::TestFixture fixture("../../../tests/clock/model.sdf");
    const int iterations = 1000;
    const int deltaTInNanoseconds = 1e6; // 1ms
    const int tolerance = 1e3; // 1 microsecond error tolerance

    yarp::os::NetworkClock netClock;
    netClock.open("/clock");

    // ACT
    fixture.Server()->Run(/*_blocking=*/true, iterations, /*_paused=*/false);
    std::this_thread::sleep_for(std::chrono::milliseconds(1000));

    // ASSERT
    double clockElapsedTimeInNanoseconds = netClock.now() * 1e9;
    double expectedSimTimeInNanoseconds = (iterations - 1) * deltaTInNanoseconds;

    ASSERT_NEAR(clockElapsedTimeInNanoseconds, expectedSimTimeInNanoseconds, tolerance);
}

```

**Expected behavior**
NetworkClock now() method returning the appropriate timestamp.

**Screenshots**

**Configuration (please complete the following information):**
 - OS: Ubuntu 22.04
 - yarp version: 3.9.0
-  compiler: conda installation

xela-95, 2024-03-20:
CC @traversaro

traversaro, 2024-03-20:
I suspect this is related to the use of NestedContact in https://github.com/robotology/yarp/blob/2e513a64f2ffaa43b91c212c4bfbf20049b576b7/src/libYARP_os/src/yarp/os/NetworkClock.cpp#L153 .

traversaro, 2024-03-22:
I added the test in https://github.com/robotology/yarp/pull/3098, and I did not reproduce the error. I think the problem in your example @xela-95 is due to something else. At the line in which you are creating calling `netClock.open("/clock");`, the simulation never run, and so the `/clock` port was never created. If you change that to:

~~~
fixture.Server()->Run(/*_blocking=*/true, iterations/2, /*_paused=*/false);
std::this_thread::sleep_for(std::chrono::milliseconds(10));
yarp::os::NetworkClock netClock;
netClock.open("/clock");
fixture.Server()->Run(/*_blocking=*/true, iterations/2, /*_paused=*/false);
std::this_thread::sleep_for(std::chrono::milliseconds(10));
~~~
 
I guess everything should work, as you would call `netClock.open("/clock");` once `/clock` port has indeed been opened.

xela-95, 2024-03-22:
> I added the test in #3098, and I did not reproduce the error. I think the problem in your example @xela-95 is due to something else. At the line in which you are creating calling [...]

Thank you that could be the problem! I'll let you know when I've tried it

traversaro, 2024-03-22:
I would go ahead and close this as according to https://github.com/robotology/yarp/pull/3098 there is no problem in YARP, we can always re-open if necessary.

closed 2024-03-22 by traversaro