robotology/yarp #3102: Unable to start yarpview on Xavier

S-Dafarra, 2024-04-12:
**Describe the bug**
I wanted to run ``yarpview`` on the screen of the `ergoCub` robot, which is connected to a Xavier board. I tried to launch it from ssh with the following command
```
yarpview
```
but I had the error
```
qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Aborted (core dumped)
```
Then I tried with
```
DISPLAY=:0 yarpview
```
and I had
```
QQmlApplicationEngine failed to load component
qrc:/qml/QtYARPView/main.qml:11 plugin cannot be loaded for module "robotology.yarp.view": Cannot load library /usr/local/src/robot/robotology-superbuild/build/install/lib/qt5/qml/robotology/yarp/view/libQtYARPViewPlugin.so: (/lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block)
```
Then with 
```
DISPLAY=:0 LD_PRELOAD=/usr/lib/aarch64-linux-gnu/libgomp.so.1 yarpview
```
and I had
```
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
Could not initialize GLX
Aborted (core dumped)
```
Finally with
```
QT_XCB_GL_INTEGRATION=none DISPLAY=:0 LD_PRELOAD=/usr/lib/aarch64-linux-gnu/libgomp.so.1 yarpview
```
``yarpview`` starts on the screen. Nonetheless, if I try to connect to a fake frame grabber, the visualization remains black. Interestingly, the statistics are updated, so I imagine that the image is received.

Under the suggestion of @randaz81, I installed ``mesa-utils`` and run ``DISPLAY=:0 glxgears``. This works nicely.

**To Reproduce**
Steps to reproduce the behavior:
Run
```
QT_XCB_GL_INTEGRATION=none DISPLAY=:0 LD_PRELOAD=/usr/lib/aarch64-linux-gnu/libgomp.so.1 yarpview
```
and in another terminal 
```
yarpdev --device fakeFrameGrabber --name /grabber --mode grid
```
and in a third one
```
yarp connect /grabber /yarpview/img:i
```

**Expected behavior**
``yarpview`` should visualize the image

**Configuration (please complete the following information):**
 - OS: Ubuntu 20.04.6 LTS
 - yarp version:  3.9.0
-  compiler: make

**Additional context**
Add any other context about the problem here.

cc @traversaro @Nicogene @randaz81 @carloscp3009

martinaxgloria, 2024-05-30:
Hi everyone, I faced the same issue with the Orin NX when trying to launch the yarpview. The solution proposed:

> Finally with
> 
> ```
> QT_XCB_GL_INTEGRATION=none DISPLAY=:0 LD_PRELOAD=/usr/lib/aarch64-linux-gnu/libgomp.so.1 yarpview
> ```

worked also for me but when I connected the yarpview to another yarp port, I always saw a black screen. 

@S-Dafarra did you try with conda packages? 

cc @Nicogene @randaz81

S-Dafarra, 2025-03-31:
With Jetpack 6 it works flawlessly. Closing

closed 2025-03-31 by S-Dafarra