robotology/yarp #3110: Error: Module not found YARP when i try to use it using python 3.11

16239014, 2024-06-03:
I will describe what i did from start:

**1.** I did source install in robsub envoirment unsing Conda forge. steps i follow `https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md#binary-installation` 

**2.** Then i install this example in same robsub `https://icub-tech-iit.github.io/documentation/sw_installation/check_your_installation/#check-icub` i run on gazebo.

**3.** Then in last i install **python 3.11.7** and using **spyder 5.5.4** in robsub and try to run this code `https://robotology.github.io/robotology-documentation/doc/html/icub_python_basic_motor.html`

**ERROR LOG:**

runfile('/home/sohail/Desktop/python folder for thesis/untitled0.py', wdir='/home/sohail/Desktop/python folder for thesis')
Traceback (most recent call last):

  File ~/anaconda3/envs/robsub/lib/python3.11/site-packages/spyder_kernels/py3compat.py:356 in compat_exec
    exec(code, globals, locals)

  File ~/Desktop/python folder for thesis/untitled0.py:2
    import yarp

ModuleNotFoundError: No module named 'yarp'

traversaro, 2024-06-03:
If you type `python -c "import yarp"` from the command line in which the `robsub` environment is activated and in which you sourced the `setup.sh` script of the robotology-superbuild what happens? If that works, the problem is probably spyder-specific and it may be more useful to ask support in spyder-related channels.

16239014, 2024-06-04:
i did what you told `python -c  "import yarp"` but i get same error **no module name yarp**

you can see step i follow maybe i did something wrong

**Error log:**
~~~
(base) sohail@sohail-Lenovo-V14-G3-IAP:~$ conda activate robsub
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~$ source /home/sohail/robotology-superbuild/build/install/share/robotology-superbuild/setup.sh
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~$ python -c "import yarp"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'yarp'

~~~

traversaro, 2024-06-04:
Thanks, can you report the content of `PYTHONPATH` env variable after you call `source /home/sohail/robotology-superbuild/build/install/share/robotology-superbuild/setup.sh`, i.e. : 

~~~
echo $PYTHONPATH
~~~

Are you sure that you enabled the `ROBOTOLOGY_USES_PYTHON=ON` CMake option of the robotology-superbuild?

16239014, 2024-06-04:
content of pyhtonpath:
~~~
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~$ echo $PYTHONPATH
/home/sohail/robotology-superbuild/build/install/lib/python3.11/site-packages
~~~
> Are you sure that you enabled the `ROBOTOLOGY_USES_PYTHON=ON` CMake option of the robotology-superbuild?

Yes i enable this

traversaro, 2024-06-04:
Can you check the content of `/home/sohail/robotology-superbuild/build/install/lib/python3.11/site-packages` and of `/home/sohail/robotology-superbuild/build/install/lib/python3.11/dist-packages` ?

16239014, 2024-06-04:
> Can you check the content of `/home/sohail/robotology-superbuild/build/install/lib/python3.11/site-packages` and of `/home/sohail/robotology-superbuild/build/install/lib/python3.11/dist-packages` ?

there no folder in **lib** named as python3.11.

traversaro, 2024-06-04:
Interesting, can you report the `/home/sohail/robotology-superbuild/build/src/YARP/install_manifest.txt` and `/home/sohail/robotology-superbuild/build/src/YARP/CMakeCache.txt` files? Thanks!

16239014, 2024-06-05:
Those files you asked for were too large, so I attached them:

> Interesting, can you report the `/home/sohail/robotology-superbuild/build/src/YARP/install_manifest.txt` 
[install_manifest.txt](https://github.com/user-attachments/files/15573064/install_manifest.txt)

`/home/sohail/robotology-superbuild/build/src/YARP/CMakeCache.txt` 
[CMakeCache.txt](https://github.com/user-attachments/files/15573066/CMakeCache.txt)

traversaro, 2024-06-05:
Did you enable the ROBOTOLOGY_USES_PYTHON superbuild option?

traversaro, 2024-06-05:
That is quite strange, in YARP's `CMakeCache.txt` you have `CREATE_PYTHON:INTERNAL=OFF`, but that variable should be set to `ON`  by robotology-superbuild. Can you run `make YARP` in the build of the superbuild and try again? Something really strange is happening.

16239014, 2024-06-05:
After run` make `in build of superbuild i found folder in lib. I know you asked for` make yarp` 

Output: 
~~~
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~/robotology-superbuild/build$ make yarp
make: *** No rule to make target 'yarp'.  Stop.
~~~
> can you report the `/home/sohail/robotology-superbuild/build/src/YARP/install_manifest.txt` and `/home/sohail/robotology-superbuild/build/src/YARP/CMakeCache.txt` files?

I DONT KNOW HOW TO REPORT IT SO ATTACH SCREENSHOT
[image]

16239014, 2024-06-11:
@traversaro Please guide me through this issue.

16239014, 2024-06-14:
@traversaro i think you forgot to reply??

traversaro, 2024-06-14:
Sorry, I asked to run `make YARP`, not `make yarp`, can you try and report the output?

16239014, 2024-06-23:
> Sorry, I asked to run `make YARP`, not `make yarp`, can you try and report the output?

sorry for late response, Build target YARP 100%
but still error in spyder occur: module not found error

traversaro, 2024-06-23:
Ok, can you please report the `CMakeCache.txt` and `install_manifest.txt` after you run `make YARP` . `python -c "import yarp"` still fails?

16239014, 2024-06-24:
> Ok, can you please report the `CMakeCache.txt` and `install_manifest.txt` after you run `make YARP`
**Files Are:**
    [CMakeCache.txt](https://github.com/user-attachments/files/15950437/CMakeCache.txt)
      [install_manifest.txt](https://github.com/user-attachments/files/15950439/install_manifest.txt)

>`python -c "import yarp"` still fails? Let's try to fix this before trying to get it to work in spyder.
 1. First i activated the environment.
 2. then source it
 3. run yarpserver
 4. In new terminal again i activated environment and sourcing then i run the command you said `python -c "import yarp` **it stuck here for more than 30 minutes**

[image]

traversaro, 2024-06-24:
You need to run `python -c "import yarp"`, not `python -c "import yarp`.

16239014, 2024-06-24:
> You need to run `python -c "import yarp"`, not `python -c "import yarp`.

**With sourcing nothing happen it just go to next line**

**Without sourcing it show error:**
```(base) sohail@sohail-Lenovo-V14-G3-IAP:~$ conda activate robsub
(robsub) sohail@sohail-Lenovo-V14-G3-IAP:~$ python -c "import yarp"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'yarp'
```

traversaro, 2024-06-24:
If you do not get any error, it means that yarp is found correctly. I am not sure how to make Spyder use conda's yarp.

traversaro, 2024-06-24:
Looking on Google, you could:
* Make sure that you start `spyder` from the `robsub` environment and in a terminal in which you sourced the `setup.sh` of the `robotology-superbuild`
* Make sure that spyder is using the python executable of the `robsub` environment, see https://docs.spyder-ide.org/current/faq.html#using-existing-environment for more details. I guess it should be `~/anaconda3/envs/robsub/bin/python` .

closed 2025-01-22 by randaz81