robotology/yarp #3116: YARP 3.9: compatibility with python 3.12 when handling images

MissingSignal, 2024-06-20:
**Describe the bug**
Dear support, 
I get a segmentation fault when handling images on python 3.12.
In particular something goes wrong when I wrap images with **_yarp_image.setExternal()_** .
The same code runs well on python < 3.11 .

**To Reproduce**
I've experienced the error on my personal code but here's a snippet to reproduce the error.

```
# Create the array with random data
img_array = numpy.random.uniform(0., 255., (240, 320)).astype(numpy.float32)
 
# Create the yarp image and wrap it around the array  
yarp_image = yarp.ImageFloat()
yarp_image.setExternal(img_array, img_array.shape[1], img_array.shape[0])
```

**Configuration (please complete the following information):**
 - OS: Ubuntu 24.04 - MacOS Sonoma 14.5
 - yarp version: latest stable release of YARP 3.9

traversaro, 2024-06-20:
Interestingly there have been some related issue but for different pairs of Python versions:
* https://github.com/robotology/yarp/issues/1873
* https://github.com/robotology/yarp/issues/1873

Perhaps we are all barking at the wrong tree, and the issue is not the Python version? To have a full picture about the problem, I think it is important to know:
* Which Python are you using? The one from apt or the one from conda? Are you sure you are recompiling YARP against the correct version of Python you want to use?
* Are you able to reproduce the problem in a clean system, for example in a Docker image?

traversaro, 2024-06-20:
By the way, https://github.com/robotology/yarp/issues/1873 suggests that you should pass `img_array.data` to setExternal instead of `img_array`, probably it may be worth to try that.

traversaro, 2024-06-20:
> By the way, #1873 suggests that you should pass `img_array.data` to setExternal instead of `img_array`, probably it may be worth to try that.

If that works, it may be worth updating the example in https://github.com/robotology/yarp/blob/bde9cd2b1069d207a77a845935b2c3063c74c58d/bindings/python/examples/yarpImage.py#L32 .

MissingSignal, 2024-06-20:
Dear Silvio,
thanks for the incredibly fast answer, I'have done some tests and you solved my error !
**We need to pass `img_array.data` to setExternal instead of `img_array`.**
As expected, this way of passing the image seems to be backwards compatible and work with earlier versions of python, I tested 3.11.

For reference I attach a working version of my previous snippet:
```
# Create the array with random data
img_array = np.random.uniform(0., 255., (240, 320)).astype(np.float32)
 
# Create the yarp image and wrap it around the array  
yarp_image = yarp.ImageFloat()
yarp_image.setExternal(img_array.data, img_array.shape[1], img_array.shape[0])
```

I was experiencing that error on:
- A clean installation of Ubuntu 24.04 + source installation of Robobotlogy Superbuild (bindings compiled against its native python 3.12)
- A virtual env created with conda, here yarp was installed through conda-forge with python 3.12

I does not seem to be dependent on any particular configuration of mine.

P.S: It may be worth updating also the example in [icub-tutorials](https://github.com/robotology/icub-tutorials/blob/master/python/python_imaging.py)

randaz81, 2025-09-01:
Probably fixed (PS: yarp 3.9 is not supported anymore)