robotology/yarp #3119: Unable to set the joint torque feedforward parameter from the motorgui in the compliant mode

GiulioRomualdi, 2024-06-26:
**Describe the bug**
I'm trying to set the torque feedforward parameter in the compliant mode tab. However the the parameter seems to be not propagated to the low level

**To Reproduce**
Steps to reproduce the behavior: This video should explain better what is happening

**Expected behavior**
The value should be sent to the low level 

**Configuration (please complete the following information):**
 - OS:
 - yarp version:
-  compiler:

**Additional context**
Add any other context about the problem here.

GiulioRomualdi, 2024-06-26:
I don't know if the problem is due to yarp, icub-main or the firmware. I opened the issue here as entry point since I spot the problem from the motorgui 

cc @isorrentino @pattacini @traversaro

pattacini, 2024-06-26:
cc @randaz81 @Nicogene @valegagge @marcoaccame

randaz81, 2025-03-23:
Any update? Can I close this? @valegagge @GiulioRomualdi

valegagge, 2025-03-25:
Sorry, but I don't remember anything... :(

@GiulioRomualdi: Do you have any new news on this issue? Would you be able to check if it is still there? If you need my help, please let me know.

GiulioRomualdi, 2025-07-02:
I completely forgot the issue: I can double-check if it is still there!

randaz81, 2025-08-21:
Task complete. Fixed by #3274