robotology/yarp #3122: Error: module 'yarp' has no attribute 'Network'

16239014, 2024-07-25:
I am running an example from [https://robotology.github.io/robotology-documentation/doc/html/icub_python_basic_motor.html](url) and when I try to initialize a Yarp network it shows an error.

ERROR LOG IN SPYDER:

~~~runfile('/home/sohail/Desktop/python folder for thesis/import yarp.py', wdir='/home/sohail/Desktop/python folder for thesis')
Traceback (most recent call last):

  File ~/anaconda3/envs/robsub/lib/python3.11/site-packages/spyder_kernels/py3compat.py:356 in compat_exec
    exec(code, globals, locals)

  File ~/Desktop/python folder for thesis/import yarp.py:3
    yarp.Network.init()

AttributeError: module 'yarp' has no attribute 'Network'

traversaro, 2024-07-25:
How did you installed YARP? Is this working if you run it outside of spyder?

16239014, 2024-07-26:
@traversaro i installed yarp with conda forge. outside spider when I do python -c "import yarp" it just go to next line and I think its fine.
so now when I use yarp in spyder It shows that yarp is imported but no attribute name as Network.

**code:**

import yarp
if hasattr(yarp, 'Network'):
    yarp.Network.init()
    
 **Error log:**

runfile('/home/sohail/Desktop/python folder for thesis/import yarp.py', wdir='/home/sohail/Desktop/python folder for thesis')
Traceback (most recent call last):

  File ~/anaconda3/envs/robsub/lib/python3.11/site-packages/spyder_kernels/py3compat.py:356 in compat_exec
    exec(code, globals, locals)

  File ~/Desktop/python folder for thesis/import yarp.py:1
    from yarp import network

ImportError: cannot import name 'network' from 'yarp' (/home/sohail/anaconda3/envs/robsub/lib/python3.11/site-packages/yarp/__init__.py)

traversaro, 2024-07-26:
Sorry, it is not clear to me if `yarp.Network.init()` works outside of spyder, by just launching the `python` interpreter.

closed 2024-07-31 by 16239014