robotology/yarp #3152: Cannot read data from RealSense camera using `yarpdev`

xela-95, 2024-12-03:
**Describe the bug**
I was following a tutorial from @nicktrem on how to stream data from the RealSense T265 camera and the command he used to launch was:

```bash
yarpdev --device multipleanalogsensorsserver --name /t256 --period 10 --subdevice realsense2Tracking
```

and then reading the raw data with `yarp read ... /t256/measures:o`

When I launched it I got the following error:
```
[ERROR] |yarp.dev.Drivers| The syntax:
[ERROR] |yarp.dev.Drivers| yarpdev --device multipleanalogsensorsserver --subdevice realsense2Tracking
[ERROR] |yarp.dev.Drivers| has been deprecated.
[ERROR] |yarp.dev.Drivers| Use the following command line instead:
[ERROR] |yarp.dev.Drivers| yarpdev --device deviceBundler --wrapper_device multipleanalogsensorsserver --attached_device realsense2Tracking
```
I think that is related to https://github.com/robotology/yarp/discussions/3078.

Then, following the suggestions I executed:
```bash
yarpdev --device deviceBundler --wrapper_device multipleanalogsensorsserver --attached_device realsense2Tracking --name /t256 --period 10
```

Getting the following error:

```
[DEBUG] |yarp.dev.PolyDriver|deviceBundler| Parameters are (attached_device realsense2Tracking) (device deviceBundler) (name "/t256") (period 10) (single_threaded 1) (wrapper_device multipleanalogsensorsserver) (wrapping_enabled "1")
[INFO] |yarp.device.DeviceBundler| Parameter 'wrapper_device' using value: multipleanalogsensorsserver
[INFO] |yarp.device.DeviceBundler| Parameter 'attached_device' using value: realsense2Tracking
[INFO] |yarp.device.DeviceBundler| Parameter 'doNotAttach' using DEFAULT value: false
[INFO] |yarp.device.DeviceBundler| Opening device: multipleanalogsensorsserver
[DEBUG] |yarp.dev.PolyDriver|multipleanalogsensorsserver| Parameters are (device multipleanalogsensorsserver) (name "/t256") (period 10) (single_threaded 1)
[INFO] |yarp.device.MultipleAnalogSensorsServer| Parameter 'name' using value: /t256
[INFO] |yarp.device.MultipleAnalogSensorsServer| Parameter 'period' using value: 10
[INFO] |yarp.dev.PolyDriver|multipleanalogsensorsserver| Created wrapper <multipleanalogsensorsserver>. See C++ class MultipleAnalogSensorsServer for documentation.
[INFO] |yarp.device.DeviceBundler| Opening device: realsense2Tracking
[DEBUG] |yarp.dev.PolyDriver|realsense2Tracking| Parameters are (device realsense2Tracking) (name "/t256") (period 10) (single_threaded 1)
[WARNING] |yarp.device.realsense2Tracking| This software module is experimental.
[WARNING] |yarp.device.realsense2Tracking| It is provided with uncomplete documentation and it may be modified/renamed/removed without any notice.
[INFO] |yarp.dev.PolyDriver|realsense2Tracking| Created device <realsense2Tracking>. See C++ class realsense2Tracking for documentation.
[INFO] |yarp.device.DeviceBundler| Attaching devices multipleanalogsensorsserver and realsense2Tracking.
[ERROR] |yarp.device.DeviceBundler| Unable to open iWrapper interface. Maybe multipleanalogsensorsserver is not a wrapper device?
[ERROR] |yarp.dev.PolyDriver|deviceBundler| Driver <deviceBundler> was found but could not open
[DEBUG] |yarp.device.multipleanalogsensorsserver| Detach complete
[ERROR] |yarp.dev.Drivers|deviceBundler| yarpdev: ***ERROR*** device not available.
[INFO] |yarp.dev.Drivers|deviceBundler| + Suggestions:
[INFO] |yarp.dev.Drivers|deviceBundler| deviceBundler
[INFO] |yarp.dev.Drivers|deviceBundler| + Do "yarpdev --list" to see list of supported devices.
```

**Configuration (please complete the following information):**
 - OS: Ubuntu 24.04 LTS
 - yarp version:  3.10.0+14-20241119.2+gitb4811f61f (on the branch of PR #3149 )
-  compiler: G++ 13.3.0

randaz81, 2025-01-22:
It looks to me that `multipleanalogsensorsserver` should be a `yarp:::dev::IWrapper` and not a `yarp::dev::IMultipleWrapper`, as indicated by its own source code:

https://github.com/robotology/yarp/blob/916b4e0c3b37ff269d8311592aa93c3e952222ed/src/devices/networkWrappers/multipleanalogsensorsserver/MultipleAnalogSensorsServer.cpp#L368-L381

@traversaro is it ok to change it?

traversaro, 2025-01-22:
Yes, I think there is no problem in inheriting from IWrapper .

closed 2025-01-22 by randaz81