robotology/yarp #3192: yInfor() bug with clang++

JiahangWu, 2025-03-14:
**Describe the bug**
There is a bug for ```yInfor()``` when I use Yarp on the M1 Mac. It seems that ```double delete stream``` issue with the ```clang++``` compiler. 

**To Reproduce**
```mamba install yarp -c conda-forge```
Just simply use ```yInfo()``` function.

**Screenshots**
The error message is:
```
[INFO] Loaded YARP-style parameters
=================================================================
==56021==ERROR: AddressSanitizer: attempting free on address which was not malloc()-ed: 0x000102885668 in thread T0
    #0 0x103c302d4 in _ZdlPv+0x74 (libclang_rt.asan_osx_dynamic.dylib:arm64e+0x642d4)
    #1 0x190c39488 in std::__1::locale::~locale()+0x58 (libc++.1.dylib:arm64e+0x38488)
    #2 0x190c255ec in std::__1::basic_streambuf<char, std::__1::char_traits<char>>::~basic_streambuf()+0x30 (libc++.1.dylib:arm64e+0x245ec)
    #3 0x10003f7bc in yarp::os::LogStream::~LogStream() LogStream.h:127
    #4 0x100035134 in stereowrapper::initialise(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, double, double, double, double, double) stereo_common.h
    #5 0x10002da1c in main stereo-offline.cpp:63
    #6 0x19096c270  (<unknown module>)

Address 0x000102885668 is a wild pointer inside of access range of size 0x000000000001.
SUMMARY: AddressSanitizer: bad-free (libclang_rt.asan_osx_dynamic.dylib:arm64e+0x642d4) in _ZdlPv+0x74
==56021==ABORTING
```

[image]

**Configuration (please complete the following information):**
 - OS: M1 Mac
 - yarp version: yarp 3.11.2
-  compiler: clang++

**Additional context**
To rough fix this issue, I simply remove one ```delete stream```, like:

[image]

It's probably not a good way to fix this issue.

traversaro, 2025-03-14:
Hello @JiahangWu, thanks for reporting the problem. As you are using `mamba`, can you report your `mamba list` and the specific example for " Just simply use yInfo() function." that is reproducing the problem. The error seems to report that you were using the AddressSanitizer?

randaz81, 2025-03-17:
Do you mean yInfo() (without r)? That function is tested by the CI, which does not report any error.. please provide your example so we can reproduce it.

Interestingly google provides a strange result on this topic (maybe a scam attempt?)

[image]

JiahangWu, 2025-03-17:
Sorry, I mistook spelling. I will provide an example soon.

JiahangWu, 2025-03-19:
Sorry, I cannot reproduce the bug again. I am going to close this and reopen it again if I meet the same bug.

closed 2025-03-19 by JiahangWu