robotology/yarp #3193: Support Yocto

vmartinlac, 2025-03-14:
**Is your feature request related to a problem? Please describe.**
We would like to use YARP on a Yocto platform. We have to write the recipe from scratch and it is inconvenient.

**Describe the solution you'd like**
An official Yocto recipe for YARP would be great.

traversaro, 2025-03-14:
Hello @vmartinlac, YARP is quite a common CMake project, so I would expect its recipe to be quite similar to any CMake-based project you compile in yocto. 

The most inconvenient part may be to make sure that also its dependencies are available in yocto, but you can optionally disable the ACE dependency with the `SKIP_ACE` CMake option.

In a nutshell, can you attemp to write a recipe, and report back if there are any problems?

vmartinlac, 2025-03-14:
I have written a recipe. ACE is supported by Yocto so it was not a problem.
I had to patch some CMake listings so that all the libraries are installed
with version number, otherwise Yocto Quality Analysis (QA) complains. Also,
I had to specify all the installed files in the recipe since they were not
specified as expected by Yocto in the CMake listing.

vmartinlac, 2025-03-14:
Also, when cross compiling, cmake try_run commands are not executed, so the result must be given to cmake. Otherwise a file was not configured correctly and there was a compilation error.

Here is the recipe. This is crude quick and dirty work to be improved.
[recipe.zip](https://github.com/user-attachments/files/19251470/recipe.zip)

randaz81, 2025-03-17:
Maybe you want to open a discussion on this topic? I have no idea about what Yocto is