robotology/yarp #3197: Bad name for yarp_priv_tinyxmlConfig.cmake

vmartinlac, 2025-03-21:
**Describe the bug**
**To Reproduce**
I clone yarp git repository, I checkout tag v3.9.1, I configure with cmake, using Ninja, with BUILD_SHARED_LIBS=OFF, I build, I install.
Then I configure my project but CMake complains that it can find neither YARP_priv_TinyXMLConfig.cmake nor yarp_priv_tinyxml-config.cmake. The actual name of the installed file is YARP_priv_tinyxmlConfig.cmake . This problem may be silent on Windows due to case insensitivity.

**Expected behavior**
The name of the installed YARP_priv_TinyXML cmake config file should either be YARP_priv_TinyXMLConfig.cmake or yarp_priv_tinyxml-config.cmake.

**Configuration (please complete the following information):**
 - OS: Ubuntu 24.04.2 LTS
 - yarp version: 3.9.1
-  compiler: gcc 13.3.0

traversaro, 2025-03-21:
Thanks, this is related to https://github.com/robotology/yarp/issues/2512, but your insight on the wrong file name probably may hint to the solution, thanks!

vmartinlac, 2025-03-21:
If I manually rename the installed files (I change the case). Then my project finds YARP_priv_TinyXML and builds and runs alright.

closed 2025-03-23 by randaz81