robotology/yarp #3198: Plugins not loading with static build

vmartinlac, 2025-03-21:
**Describe the bug**
I configure with BUILD_SHARED_LIBS=OFF and YARP_FORCE_DYNAMIC_PLUGINS=OFF .
I compile, I build my project and I run my executable.

- I notably see no shmem carrier in the yarpserver output so I investigate.
- In function, yarpCustomInit, function yarpcar_INIT_FUNCTION is never called.

In CMake listing yarp/src/libYARP_init/src/CMakeLists.txt line 42 (commit v3.9.1) there is a reference to file CustomInit.cpp which does not exist. The actual file is Network.CustomInit.cpp.
If I replace that with 

`target_compile_definitions(YARP_init PRIVATE yarpcar_INIT_FUNCTION=add_yarpcar_plugins)`

Then the plugins are correctly initialized.

So the CMake listing seems to add definitions to an unexisting file so the actual file does not get the symbols so the plugins are not initialized and I have no shmem carrier.

**Configuration (please complete the following information):**
 - OS: Ubuntu 24.04.2
 - yarp version: v3.9.1
-  compiler: gcc 13.3.0

traversaro, 2025-03-21:
Thanks for the feedback. I suspect that the support for full static builds (including plugins) was there working in the past, but it is not covered by tests or used a lot, so something broke over time.

randaz81, 2025-03-22:
Definitely we need to add a test in the CI

randaz81, 2025-03-24:
Issue confirmed by https://github.com/robotology/yarp/pull/3202

I'm not sure about what you mean here @vmartinlac
```
not exist. The actual file is Network.CustomInit.cpp.
If I replace that with
target_compile_definitions(YARP_init PRIVATE yarpcar_INIT_FUNCTION=add_yarpcar_plugins)
Then the plugins are correctly initialized.
```
Have you already found a fix? Can you propose a PR?

vmartinlac, 2025-03-24:
A fix is replacing

```
 set_property(
      SOURCE CustomInit.cpp
      APPEND PROPERTY COMPILE_DEFINITIONS yarpcar_INIT_FUNCTION=add_yarpcar_plugins
    )
```

with 

`target_compile_definitions(YARP_init PRIVATE yarpcar_INIT_FUNCTION=add_yarpcar_plugins)`

In file src/libYARP_init/src/CMakeLists.txt

traversaro, 2025-03-24:
Thanks @vmartinlac ! I think a more minimal change (that is more inline with the original goal of the code) is to just fix the value passed to `set_property(SOURCE` . The `CustomInit.cpp` was renamed in `Network.CustomInit.cpp` in https://github.com/robotology/yarp/commit/3eb97000f57d388bc746e7304eb46de1c2c0e1b0#diff-e26e6136eee05544fca0d9360616145d09f0910dff53c12cbd8c046c0790fb5a , but the `CustomInit.cpp` value specified in the CMake code was not updated.

traversaro, 2025-03-24:
> Thanks [@vmartinlac](https://github.com/vmartinlac) ! I think a more minimal change (that is more inline with the original goal of the code) is to just fix the value passed to `set_property(SOURCE` . The `CustomInit.cpp` was renamed in `Network.CustomInit.cpp` in [3eb9700#diff-e26e6136eee05544fca0d9360616145d09f0910dff53c12cbd8c046c0790fb5a](https://github.com/robotology/yarp/commit/3eb97000f57d388bc746e7304eb46de1c2c0e1b0#diff-e26e6136eee05544fca0d9360616145d09f0910dff53c12cbd8c046c0790fb5a) , but the `CustomInit.cpp` value specified in the CMake code was not updated.

Fix proposed in https://github.com/robotology/yarp/pull/3209 .

randaz81, 2025-03-24:
Merged thanks! There are still a bunch of errors due to multiple defined symbols in the output static library, but we can manage these easily. I'll fix them in a separate PR.

randaz81, 2025-03-26:
Fixed.

closed 2025-03-26 by randaz81