robotology/yarp #3224: Missing python bindings for IFrameTransform

steb6, 2025-05-20:
While I was trying to use python bindings for IFrameTransform, I got the following error:

```
Traceback (most recent call last):
  File "C:\Users\sberti\metaCub_software\metaCub\test.py", line 67, in <module>
    tester.test_get_transform("/root", "/hand")  # Replace with actual frame names
    ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
  File "C:\Users\sberti\metaCub_software\metaCub\test.py", line 47, in test_get_transform
    success = self.tf_reader.getTransform(src_frame, dst_frame, None)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'SwigPyObject' object has no attribute 'getTransform'
```

The feature has been added in #3162 .
I leave you a minimal code snipper to reproduce the problem

```
conda create -n test_bindings yarp==3.11.2
conda activate test_bindings
- yarpserver
- yarpdev --device frameTransformServer --filexml_option fts_yarp_only.xml
- python test.py
```
test.py:
```
import yarp
import time

# Initialize YARP
yarp.Network.init()

class TFClientTest:
    def __init__(self):
        self.tf_driver = yarp.PolyDriver()
        self.tf_reader = None
        self.local_port_prefix = "test_tf_client"

    def setup(self):
        # Simulate ResourceFinder values
        tf_client_cfg = yarp.Property()
        tf_client_cfg.put("device", "frameTransformClient")
        tf_client_cfg.put("filexml_option", "ftc_yarp_only.xml")
        tf_client_cfg.put("ft_client_prefix", f"/{self.local_port_prefix}/tf")
        tf_client_cfg.put("ftc_storage_timeout", 300.0)
        tf_client_cfg.put("ft_server_prefix", "")  # Change if needed
        tf_client_cfg.put("local_rpc", f"/{self.local_port_prefix}/tf/local_rpc")

        print("[Test] Opening frameTransformClient driver...")
        if not self.tf_driver.open(tf_client_cfg):
            print(f"[ERROR] Unable to open TF polydriver with options:\n{tf_client_cfg.toString()}")
            return False

        self.tf_reader = self.tf_driver.viewIFrameTransform()
        if self.tf_reader is None:
            print("[ERROR] Unable to view IFrameTransform interface.")
            self.tf_driver.close()
            return False

        print("[Test] frameTransformClient successfully opened and IFrameTransform interface acquired.")
        return True

    def test_get_transform(self, src_frame="/root", dst_frame="/hand"):
        if not self.tf_reader:
            print("[ERROR] IFrameTransform interface not available.")
            return False
        time.sleep(1.0)
        success = self.tf_reader.getTransform(src_frame, dst_frame, None)
        print(success)

    def cleanup(self):
        if self.tf_driver.isValid():
            self.tf_driver.close()
        print("[Test] Cleanup done.")

# Run the test
if __name__ == "__main__":
    tester = TFClientTest()
    if tester.setup():
        tester.test_get_transform("/root", "/hand")  # Replace with actual frame names
    tester.cleanup()

    # Finalize YARP
    yarp.Network.fini()

```

thank you!

randaz81, 2025-05-20:
@steb6 you can edit the file `yarp.i` and add the missing functions you need, just following the super-easy example: https://github.com/robotology/yarp/pull/3162/files (line 1593)
Please feel free to open a PR when done.

steb6, 2025-05-28:
Thanks @randaz81 , however I tried to compile yarp 3.11.2 from source and to call the ```getAllFrameIds``` method, that is already defined in https://github.com/robotology/yarp/pull/3162/files, but I keep getting the error ```AttributeError: 'SwigPyObject' object has no attribute 'getAllFrameIds '``` that means that the 3 methods added in the pull that you linked are not present, and so adding another one wouln't solve the issue

randaz81, 2025-05-29:
fixed by https://github.com/robotology/yarp/pull/3231

closed 2025-05-29 by randaz81